RIOT-OS/RIOT #7152: SAUL: add full support for PhyNode (pba-d-01-kw2x)

smlng, 2017-06-07:
The following table gives an overview and work in progress on the state of SAUL support for peripherals of the PhyTec PhyNode (board `pba-d-01-kw2x`), looks quite good already 😄 

Device | Driver | SAUL support
------ | ------ | ----
Humidity Sensor | HDC1000 | ✅ 
Magnetometer | MAG3110 | ✅ 
Accelerometer | MMA8x5x | ✅ 
Pressure (+Temp) | MPL3115A2 | 🚧  see #7120 
IR Temperature | TMP006 🏗  | 🚫 
Color Light Sensor | TCS37727 | ✅ 
Capacative Sensor (Button) | periph_gpio | 🚧   see #7147 
RGB LED | periph_gpio | 🚧  see #7147

🏗 needs rework

smlng, 2017-06-29:
all done! Full SAUL support for board `pba-d-01-kw2x`! Closing this issue!

closed 2017-06-29 by smlng

smlng, 2017-06-29:
@aabadie might be something for the release notes?

aabadie, 2017-06-29:
> might be something for the release notes?

sure, I'll put this in shape once we reach feature freeze.