RIOT-OS/RIOT #7182: Question : Border router setting

kamejoko80, 2017-06-14:
Hello,

I've followed this [guideline](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_border_router) to setting up border router connection, but I am not able to ping from Linux PC to a node device. I have 3 devices: Linux PC, BR and a node. The interface of each device as below:

1> Linux PC interface

`tap0      Link encap:Ethernet  HWaddr ae:a7:d9:f3:3c:e3
          inet6 addr: fe80::aca7:d9ff:fef3:3ce3/64 Scope:Link
          inet6 addr: fe80::1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:34 errors:0 dropped:0 overruns:0 frame:0
          TX packets:72 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:2648 (2.6 KB)  TX bytes:9310 (9.3 KB)`

2> BR interface

`Iface  6   HWaddr: 11:1e  Channel: 26  Page: 0  NID: 0x23
           Long HWaddr: 7b:62:73:34:03:3e:11:1e
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
           ACK_REQ  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::7962:7334:33e:111e/64  scope: local
           inet6 addr: ff02::1:ff3e:111e/128  scope: local [multicast]
           inet6 addr: 2001:db8::7962:7334:33e:111e/64  scope: global
           inet6 addr: ff02::2/128  scope: local [multicast]

Iface  7   HWaddr: 00:21:d3:60:5a:8d

           MTU:1500  HL:64  RTR  RTR_ADV
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::221:d3ff:fe60:5a8d/64  scope: local
           inet6 addr: ff02::1:ff60:5a8d/128  scope: local [multicast]
           inet6 addr: ff02::2/128  scope: local [multicast]
           inet6 addr: fe80::2/64  scope: local
           inet6 addr: ff02::1:ff00:2/128  scope: local [multicast]`

3> Node interface

`Iface  7   HWaddr: 15:5a  Channel: 26  Page: 0  NID: 0x23
           Long HWaddr: 11:16:72:2f:11:30:15:5a
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
           ACK_REQ  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::1316:722f:1130:155a/64  scope: local
           inet6 addr: ff02::1:ff30:155a/128  scope: local [multicast]
           inet6 addr: ff02::1a/128  scope: local [multicast]
           inet6 addr: 2001:db8::1316:722f:1130:155a/64  scope: global
           inet6 addr: ff02::2/128  scope: local [multicast]

           Statistics for Layer 2
            RX packets 35  bytes 2476
            TX packets 173 (Multicast: 158)  bytes 11982
            TX succeeded 13 errors 0
           Statistics for IPv6
            RX packets 26  bytes 2608
            TX packets 173 (Multicast: 158)  bytes 11490
            TX succeeded 173 errors 0`

I was able to ping from BR (or node) to Linux PC (fe80::aca7:d9ff:fef3:3ce3) but I'm not able to ping from Linux PC to node device. Please help me how to configure the BR in order to ping from Linux to another node device ? 

I see this from the guideline but I am not clear 

`> ifconfig 6 add 2001:db8::2/48
> ifconfig 5 add 2001:db8::3/64
> fibroute add :: via <link-local of tap> dev 6`

Thank you very much.

miri64, 2017-07-15:
@kamejoko80 what board are you using as a node device? What board as a border router?

kamejoko80, 2017-07-29:
Hello miri64,

I'm using nucleo32-l432 for both node and border router. I see there are TAP and SLIP on the wiki guideline page, TAP doesn't work but SLIP is working properly for me. 

Thank you very much

miri64, 2019-01-24:
I think this should be solved by now. If you disagree, please reopen.

miri64, 2019-01-24:
(reasoning: from the shell output and time of discussion I believe this was using the old network layer infrastructure, which has been overhault completely)

closed 2019-01-24 by miri64