RIOT-OS/RIOT #7188: CC2650STK shell not reading chars from UART

A-Paul, 2017-06-14:
The *tests/shell* doesn't react to input. Starting message is coming up, so writing to stdout (UART) is working. After that there is no reaction.

nRhR, 2017-06-14:
Hey @A-Paul thanks for linking.
Im interested, how can we solve this issue?
Is there any method to figure out the bugg?
Any ideas someone ?

Kind regards

smlng, 2017-06-20:
I can confirm this issue, reading UART works, writing to it does not. But I also observed that the buffer seems not do be flushed at the end, the last line of output is always missing.

Btw. lots of tests do not work as expected on this platform, i.e. `periph_timer` runs but timers fire immediately regardless of the delay set. And as the base-timer is not working all xtimer tests fail, too.

A-Paul, 2017-06-22:
Hello @nRhR ,
generally we have to "hunt" the bug down. ;)
*shell_run()* is essentially a loop over *readline()* and  *handle_input_line()* - all of them in <sys/shell/shell.c>. *readline* is a loop that calls *getchar()*.

Hey @smlng, 
> I can confirm this issue, reading UART works, writing to it does not.

This is a external view, I think. ;) I experienced a hangup when reading the UART - from RIOTs perspective.

lcgamboa, 2017-06-26:
I can confirm the bug running the default example in ccc2650 launchpad. When I using the code composer to debug step by step until the call of shell function and run after this point the example works. Apparently the problem is the corruption between RIOT threads initialization and can be linked with timers issue.

smlng, 2017-06-26:
@lcgamboa good point! Timers are not working correctly, try `tests/periph_timer` - though it seems to work, its output is nonsense.

A-Paul, 2017-07-12:
Issue is solved by #7342

smlng, 2017-07-12:
then lets close this?

A-Paul, 2017-07-12:
@smlng, strange suggestion. ;)

closed 2017-07-12 by A-Paul