RIOT-OS/RIOT #7189: Missing support for building position independent code

gebart, 2017-06-14:
Adding support for building position independent code would make it easier to build images for online firmware updates, since the binary can be relocated by the bootloader to a suitable location depending on available memory. Running binaries from RAM should also be possible without any special linker scripts.  

I have not investigated the details but the a rough draft of the changes needed are:
- Add -fPIC to CFLAGS 
- add linker script handling of .got (global offset table) and .got.plt (procedure linkage table) 
- fill GOT during boot (I don't know if gcc provides code for this or if we need to write our own)

gebart, 2017-06-14:
Maybe relevant #6450 #7167 #6922 #7081

miri64, 2019-08-10:
@kaspar030 @kYc0o isn't this fixed via your work?

kYc0o, 2019-08-12:
Unfortunately not. This is about dynamic loading of position independent code during compile time, if I'm not mistaken. Our approach is rather to boot images at a fixed position given by a specific linker script, which the method discussed here would avoid.

closed 2020-03-15 by stale[bot]