RIOT-OS/RIOT #7190: Regression: PR #7134 makes hello-world no longer output

photonthunder, 2017-06-14:
@smlng and @kYc0o, PR #7134 compiles but there is no USART output for hello-world on samd21-xpro compiled on Mac OS X 10.12.  Removing the PR makes it function again.

travisgriggs, 2017-06-16:
as of `a6fcbcd`, my samd2-xpro prints hello

smlng, 2017-06-19:
cannot confirm this bug, after some fiddling with the flasher, I tested with `examples/hello-world` and got the following output:

```
2017-06-19 10:57:39,736 - INFO # main(): This is RIOT! (Version: 2017.04-devel-1311-ged989-MacBook-Air.lan)
2017-06-19 10:57:39,738 - INFO # Hello World!
2017-06-19 10:57:39,741 - INFO # You are running RIOT on a(n) samr21-xpro board.
2017-06-19 10:57:39,744 - INFO # This board features a(n) samd21 MCU.
```

I tested with macOS 10.12.4 and latest master (`ed9899ad`). I think we can close?

photonthunder, 2017-06-19:
I pulled the latest and it works fine. Closing.

closed 2017-06-19 by photonthunder

smlng, 2017-06-19:
Still: keep up testing and working with RIOT on macOS, it is highly appreciated 😄  And report any bugs, or inconsistencies you encounter.