RIOT-OS/RIOT #7192: gcoap confirmable messaging plan

kb2ma, 2017-06-15:
I created a [wiki page](https://github.com/kb2ma/RIOT/wiki/Confirmable-Messaging) that describes the messaging scenarios that gcoap must handle for the confirmable message type. The discussion references supporting features in gcoap, which can be seen in a [commit](https://github.com/kb2ma/RIOT/commit/52f8c4221a5d314e4660eba75f0f8538a1e40ed4) I created just for documentation. I intend to implement these features incrementally, in digestible commits. See below for the implementation plan, and of course I'm happy to hear any feedback or suggestions.

1. Create the infrastructure, and get the existing messaging to use it. This work includes:

   a. Revise gcoap_request_memo_t and add gcoap_resend_t. The hdr_buf attribute will be accessed within the union.

   b. Rename the 'observers' array to 'remotes', because gcoap will use the array to store the endpoint for a confirmable message. Also rename _find_observer() to _find_remote_ep().

2. Implement confirmable messaging with the response piggybacked on the ACK -- scenario (1) on the wiki page. This includes addition of the resend_bufs array.

3. Implement confirmable Observe notifications -- scenario (5).

4. Add gcoap_req_init_opts().

5. Implement confirmable messaging with separate responses -- scenario (2).

6. Implement synchronous messaging as described in scenario (1).

chrysn, 2020-07-01:
A few issues/PR have accumulated that should have been created as sub-tasks in here:
* [ ] #14169 (Gcoap does not accept separate responses): ready to be reviewed at #14178 

  This is an important part of a full CoAP implementations because servers may at any time start responding separately.

* [ ] #14393 (Gcoap does not send separate responses): PR draft at #14395

  This is far less urgent as the Gcoap server may just decide never to do this; relevant though for cases of resources whose representation is not available immediately (on a retransmission timer time scale; anything <0.1s is perfectly fine, and 1s just so is).

The wiki page grew a bit outdated over time (eg. w/rt `gcoap_req_send2`), I'd like to overhaul it but will need to work a bit more through what has been done already to be sure to capture all of it.

maribu, 2022-09-16:
ping?