RIOT-OS/RIOT #7196: cpu/samd21 : external interrupts are broken on samd21

TomKeddie, 2017-06-15:
I've observed on SAMD21 that allexternal interrupt controls map to EXTINT0.

Dump with no interrupts enabled.

(gdb) x/40bx 0x40001800
0x40001800: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001808: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001810: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001818: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001820: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

with EXTINT 7 enabled (note that the EXTINT0 enables are all set).

(gdb) x/40bx 0x40001800
0x40001800: 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001808: 0x01 0x00 0x00 0x00 0x01 0x00 0x00 0x00
0x40001810: 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00
0x40001818: 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00
0x40001820: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00

TomKeddie, 2017-06-15:
The problem appears to be a typo in gpio.c (I don't see any other evidence of support for the SAMR21 cpu).

> --- a/cpu/samd21/periph/gpio.c
> +++ b/cpu/samd21/periph/gpio.c
> @@ -46,7 +46,7 @@ static const int8_t exti_config[2][32] = {
>        0,  1,  2,  3,  4,  5,  6,  7, 12, 13, -1, 15,  8, -1, 10, 11},
>      { 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15,
>        0,  1, -1, -1, -1, -1,  6,  7, -1, -1, -1, -1, -1, -1, 14, 15},
> -#elif CPU_MODEL_SAMR21G18A
> +#elif CPU_MODEL_SAMD21G18A
>      {-1,  1, -1, -1,  4,  5,  6,  7, -1,  9, 10, 11, 12, 13, 14, 15,
>       -1,  1,  2,  3, -1, -1,  6,  7, 12, 13, -1, 15,  8, -1, 10, 11},
>      { 0, -1,  2,  3, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1, -1,

Will submit a pull request later today.

thomaseichinger, 2017-06-15:
@TomKeddie Could you check if https://github.com/RIOT-OS/RIOT/pull/7125 fixes this?

TomKeddie, 2017-06-16:
@thomaseichinger thanks.  This is definitely a dupe of #7125, the fix proposed there is good.  Thanks, I'm not sure why my search didn't find it.

closed 2017-06-16 by TomKeddie