RIOT-OS/RIOT #7197: boards : add support for SparkFun SAMD21 Mini Breakout

TomKeddie, 2017-06-15:
This is a tracking issue for support for the SparkFun SAMD21 Mini Breakout (https://www.sparkfun.com/products/13664).

Wiki entry is [here](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-SparkFun-SAMD21-Mini)

iia, 2018-04-30:
Hi @TomKeddie ,

I am interested to contribute for this board. What is the current state of the support for this board and what are major pending tasks ?

Cheers.

TomKeddie, 2018-04-30:
Hi @iia thanks for offering.  The work is essentially done but I am backlogged on other things and have not been able to complete it.  I think the main task would be to test and audit all the pin mappings especially the Arduino ones and the serial/spi/i2c definitions,.

My code is all checked in on https://github.com/TomKeddie/RIOT, it needs to be on a branch, my git skills are proving to be a problem (again time to focus on it).

If you can take what I've done and run with it I'd really appreciate it.

Thanks.

closed 2019-09-10 by stale[bot]