RIOT-OS/RIOT #7199: core: "Invalid read of size 4"

miri64, 2017-06-16:
I found this error with valgrind:

```
==12420== Invalid read of size 4
==12420==    at 0x8049E7E: thread_measure_stack_free (core/thread.c:134)
==12420==    by 0x804F709: ps (sys/ps/ps.c:97)
==12420==    by 0x804AC00: core_panic (core/panic.c:65)
==12420==    by 0x8049D34: _assert_failure (core/assert.c:24)
==12420==    by 0x804D0C7: _get_set (sys/net/gnrc/netapi/gnrc_netapi.c:58)
==12420==    by 0x804D2EA: gnrc_netapi_get (sys/net/gnrc/netapi/gnrc_netapi.c:161)
==12420==    by 0x8061511: _get_l2src (sys/net/gnrc/network_layer/ndp2/gnrc_ndp2.c:558)
==12420==    by 0x806106D: gnrc_ndp2_nbr_sol_send (sys/net/gnrc/network_layer/ndp2/gnrc_ndp2.c:233)
==12420==    by 0x805F08E: _probe_nbr (sys/net/gnrc/network_layer/ipv6/nib/nib.c:860)
==12420==    by 0x805ED21: _resolve_addr (sys/net/gnrc/network_layer/ipv6/nib/nib.c:744)
==12420==    by 0x805E135: gnrc_ipv6_nib_get_next_hop_l2addr (sys/net/gnrc/network_layer/ipv6/nib/nib.c:129)
==12420==    by 0x805B0BA: _send (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:799)
==12420==  Location 0x807cbcc is 0 bytes inside main_stack[10668],
==12420==  a global variable declared at kernel_init.c:78
==12420==
```

From what I was able to gather, this means that we get a bit to many bytes at the end of `main_stack`, but I'm not sure if this is a false positive.

miri64, 2017-06-16:
(marked as minor bug, since though it is in core it is just affecting `ps` with stacktest enabled [so it shouldn't be in production code anyway])

chrysn, 2022-08-25:
Can you confirm this is still here?

miri64, 2022-08-25:
Did awfully little work to document what I did there, but I will try to reproduce ASAP.

miri64, 2022-08-25:
Unable to reproduce

#### tap0

```console
$ make -C examples/gnrc_networking -j term-valgrind
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking'
/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/kconfig.mk:118: Warning! SHOULD_RUN_KCONFIG is not set but a previous configuration file was detected (did you run `make menuconfig`?). Kconfig will run regardless.
valgrind --leak-check=full --track-origins=yes --fullpath-after=/home/mlenders/Repositories/RIOT-OS/RIOT/ --read-var-info=yes /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0
==69508== Memcheck, a memory error detector
==69508== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==69508== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==69508== Command: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0
==69508== 
--69508-- WARNING: Serious error when reading debug info
--69508-- When reading debug info from /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf:
--69508-- Can't make sense of .rodata section mapping
RIOT native interrupts/signals initialized.
RIOT native board initialized.
RIOT native hardware initialization complete.

NETOPT_TX_END_IRQ not implemented by driver
main(): This is RIOT! (Version: 2022.10-devel-36-gb2a542-2022.07-branch)
RIOT network stack example application
All up, running the shell now
> ping fe80::d827:1dff:fea8:6424
ping fe80::d827:1dff:fea8:6424
12 bytes from fe80::d827:1dff:fea8:6424%6: icmp_seq=0 ttl=64 time=31.653 ms
12 bytes from fe80::d827:1dff:fea8:6424%6: icmp_seq=1 ttl=64 time=3.274 ms
12 bytes from fe80::d827:1dff:fea8:6424%6: icmp_seq=2 ttl=64 time=8.690 ms

--- fe80::d827:1dff:fea8:6424 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 3.274/14.539/31.653 ms
> udp send
udp send
usage: udp send <addr> <port> <data> [<num> [<delay in us>]]
> udp send fe80::d827:1dff:fea8:6424 1337 test 3 1000
udp send fe80::d827:1dff:fea8:6424 1337 test 3 1000
Success: sent 4 byte(s) to [fe80::d827:1dff:fea8:6424]:1337
Success: sent 4 byte(s) to [fe80::d827:1dff:fea8:6424]:1337
Success: sent 4 byte(s) to [fe80::d827:1dff:fea8:6424]:1337
> ^C
native: exiting
==69508== 
==69508== HEAP SUMMARY:
==69508==     in use at exit: 0 bytes in 0 blocks
==69508==   total heap usage: 13 allocs, 13 frees, 1,300 bytes allocated
==69508== 
==69508== All heap blocks were freed -- no leaks are possible
==69508== 
==69508== For lists of detected and suppressed errors, rerun with: -s
==69508== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

#### tap1

```console
$ PORT=tap1 make -C examples/gnrc_networking term-valgrind
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking'
/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/kconfig.mk:118: Warning! SHOULD_RUN_KCONFIG is not set but a previous configuration file was detected (did you run `make menuconfig`?). Kconfig will run regardless.
valgrind --leak-check=full --track-origins=yes --fullpath-after=/home/mlenders/Repositories/RIOT-OS/RIOT/ --read-var-info=yes /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap1
==69729== Memcheck, a memory error detector
==69729== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==69729== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==69729== Command: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap1
==69729== 
--69729-- WARNING: Serious error when reading debug info
--69729-- When reading debug info from /home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf:
--69729-- Can't make sense of .rodata section mapping
[... 79 lines cut ...]
traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
flow label: 0x00000
length: 12  next header: 17  hop limit: 64
source address: fe80::e0bc:7dff:fecb:f550
destination address: fe80::d827:1dff:fea8:6424
~~ SNIP  3 - size:  20 byte, type: NETTYPE_NETIF (-1)
if_pid: 6  rssi: -32768  lqi: 0
flags: 0x0
src_l2addr: E2:BC:7D:CB:F5:50
dst_l2addr: DA:27:1D:A8:64:24
~~ PKT    -  4 snips, total size:  72 byte
fe80::d827:1dff:fea8:6424^C
native: exiting
==69729== 
==69729== HEAP SUMMARY:
==69729==     in use at exit: 0 bytes in 0 blocks
==69729==   total heap usage: 127 allocs, 127 frees, 12,811 bytes allocated
==69729== 
==69729== All heap blocks were freed -- no leaks are possible
==69729== 
==69729== For lists of detected and suppressed errors, rerun with: -s
==69729== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

Given that @maribu et al. put a lot of work to fix unaligned access, I think we can consider this issue as fixed.

closed 2022-08-25 by miri64