RIOT-OS/RIOT #7200: edbg issues (samd21)

travisgriggs, 2017-06-16:
I only took a few weeks off... Before then, `make flash` to my samd21-xpro board used my brew installed version openocd. Life was good.

I grabbed latest riot master an hour ago, and `make flash` now tries to fetch and build `edbg`. Unfortunately it crashes and burns:

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in mac
  CC       hid.lo
hid.c:50:90: warning: unused parameter 'attr' [-Wunused-parameter]
static int pthread_barrier_init(pthread_barrier_t *barrier, const pthread_barrierattr_t *attr, unsigned int count)
                                                                                         ^
hid.c:252:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long')
      [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
hid.c:543:65: warning: unused parameter 'result' [-Wunused-parameter]
static void hid_device_removal_callback(void *context, IOReturn result,
                                                                ^
hid.c:544:47: warning: unused parameter 'sender' [-Wunused-parameter]
                                        void *sender)
                                              ^
hid.c:556:57: warning: unused parameter 'result' [-Wunused-parameter]
static void hid_report_callback(void *context, IOReturn result, void *sender,
                                                        ^
hid.c:556:71: warning: unused parameter 'sender' [-Wunused-parameter]
static void hid_report_callback(void *context, IOReturn result, void *sender,
                                                                      ^
hid.c:557:42: warning: unused parameter 'report_type' [-Wunused-parameter]
                         IOHIDReportType report_type, uint32_t report_id,
                                         ^
hid.c:557:64: warning: unused parameter 'report_id' [-Wunused-parameter]
                         IOHIDReportType report_type, uint32_t report_id,
                                                               ^
hid.c:1023:60: warning: unused parameter 'dev' [-Wunused-parameter]
int HID_API_EXPORT_CALL hid_get_indexed_string(hid_device *dev, int string_index, wchar_t *string, size_t maxlen)
                                                           ^
hid.c:1023:69: warning: unused parameter 'string_index' [-Wunused-parameter]
int HID_API_EXPORT_CALL hid_get_indexed_string(hid_device *dev, int string_index, wchar_t *string, size_t maxlen)
                                                                    ^
hid.c:1023:92: warning: unused parameter 'string' [-Wunused-parameter]
int HID_API_EXPORT_CALL hid_get_indexed_string(hid_device *dev, int string_index, wchar_t *string, size_t maxlen)
                                                                                           ^
hid.c:1023:107: warning: unused parameter 'maxlen' [-Wunused-parameter]
int HID_API_EXPORT_CALL hid_get_indexed_string(hid_device *dev, int string_index, wchar_t *string, size_t maxlen)
                                                                                                          ^
hid.c:1031:68: warning: unused parameter 'dev' [-Wunused-parameter]
HID_API_EXPORT const wchar_t * HID_API_CALL  hid_error(hid_device *dev)
                                                                   ^
13 warnings generated.
  CCLD     libhidapi.la
clang: warning: argument unused during compilation: '-pthread' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-pthread' [-Wunused-command-line-argument]
Making all in hidtest
  CXX      hidtest.o
  CXXLD    hidtest
arm-none-eabi-g++: error: CoreFoundation: No such file or directory
arm-none-eabi-g++: error: IOKit: No such file or directory
arm-none-eabi-g++: error: unrecognized command line option '-framework'
arm-none-eabi-g++: error: unrecognized command line option '-framework'
arm-none-eabi-g++: error: unrecognized command line option '-pthread'
make[5]: *** [hidtest] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [hidapi/mac/.libs/libhidapi.a] Error 2
make[1]: *** [all] Error 2
make: *** [/Users/travisg/Projects/riotRIOT/dist/tools/edbg/edbg] Error 2
```

I have no idea why this is. And am only slightly cranky that it has regressed. :/

emmanuelsearch, 2017-06-22:
I ran into the same issue on my Mac. I assume you are on OS X, right?

kaspar030, 2017-06-22:
@travisgriggs You should be able to get back to using openocd by setting "PROGRAMMER=openocd".
Sorry for the regression...

kaspar030, 2017-06-22:
> You should be able to get back to using openocd by setting "PROGRAMMER=openocd".

We could make openocd the default on OSX machines, until we can fix edbg's build...

smlng, 2017-06-22:
I can confirm this issue on macOS, and I can also confirm that @kaspar030 workaround works, I tried some days ago.

thomaseichinger, 2017-06-22:
It is actually only compilation in the `hidtest` directory that fails. `edbg` gets built successfully on the second run of `make flash`.

For some weird reason hidtest's configure thinks `CXX=arm-none-eabi-g++` is a good idea ...

kYc0o, 2017-06-26:
I confirm this behaviour and now I have edbg working on my OSX system. Should we close the issue or try to make it compile in the first try?

smlng, 2017-06-26:
I'd say the latter would be great?!

kYc0o, 2017-06-26:
Yeah well would be great but I have no idea on how to fix it, and since there's nothing to do but just type `make flash` twice that's why I propose to close the issue, but let's leave it open until only one `make flash` is necessary.

smlng, 2017-06-26:
> no idea on how to fix it
 me neither 😞  Not sure if we can unset the CXX variable when compiling edbg and reset it to the RIOT setting afterwards?

smlng, 2017-06-30:
@travisgriggs please try #7283 and report back, whether it helps or not.

PR #7283 (2017-07-04): tools, edbg: fix compiler issue on macos
fixes #7200 

simply unsetting the compiler vars (CC, CXX) helps, at least for me.
closed 2017-07-04 by thomaseichinger