RIOT-OS/RIOT #7226: edbg: tool fails to automatically build

emmanuelsearch, 2017-06-22:
trying to flash a SAMR21 on via Vagrant (Ubuntu 16.04.2 LTS) I get the following error while (automagically) building edbg before flashing:

```
[INFO] edbg binary not found - building it from source now
CC= CFLAGS= make -C /home/vagrant/RIOT/dist/tools/edbg
"make" -C /home/vagrant/RIOT/dist/tools/edbg/bin
gcc  -W -Wall -Wextra -O2 -std=gnu99 dap.c edbg.c target.c target_atmel_cm0p.c target_atmel_cm3.c target_atmel_cm4.c target_atmel_cm7.c dbg_lin.c -ludev -o edbg
dbg_lin.c:42:21: fatal error: libudev.h: No such file or directory
compilation terminated.
```

miri64, 2017-06-22:
You need to update your vagrant image. This was solved in https://github.com/RIOT-OS/RIOT/pull/7105

miri64, 2017-06-22:
[For future references](https://github.com/RIOT-OS/RIOT/pulls?utf8=%E2%9C%93&q=libudev-dev) ;-)

cgundogan, 2017-06-22:
@emmanuelsearch can you try if `vagrant box update` is enough to resolve this issue for you?

emmanuelsearch, 2017-06-22:
I had already done the update. Vagrant box update gives currently: 

```
$ vagrant box update
==> default: Checking for updates to 'RIOT/ubuntu1604'
    default: Latest installed version: 0.0.2
    default: Version constraints: 
    default: Provider: virtualbox
==> default: Box 'RIOT/ubuntu1604' (v0.0.2) is running the latest version.
```

emmanuelsearch, 2017-06-22:
(hence: reopened the issue)

cgundogan, 2017-06-22:
hm, then a complete destroy might help: `vagrant halt` `vagrant destroy` `vagrant up`

cgundogan, 2017-06-22:
oh and in case there are some older versions around: could you open the virtualbox GUI and delete them manually?

emmanuelsearch, 2017-06-22:
OK, with this it works thanks ;)

closed 2017-06-22 by emmanuelsearch