RIOT-OS/RIOT #7233: Incorrect behavior of fopen/fclose/... functions on native

vincent-d, 2017-06-23:
While playing with files, we realized that fopen/fclose/... functions on native do not use the vfs, even if open/close/read/... do.

I guess this is because RIOT is linked with the native libc, but I wonder how we could make these calls use the RIOT syscalls.

Does anyone have an idea about that?

gebart, 2017-06-23:
Could it be possible to build newlib for the native platform? It would mean one more manual installation step for the native platform, but it could perhaps be made optional through the Makefiles.

smlng, 2017-07-13:
so is this a problem/bug and do we need to (or should) address this? At least the issue title is somewhat misleading, as it doesn't seem to be a bug.

gebart, 2017-07-13:
I think it should be changed, at least If using the VFS module on native. It is confusing behavior that fopen  and open are working with different sets of files (file systems).

gebart, 2017-07-13:
OTOH it may be useful to be able to access the regular file system from inside RIOT in native.

smlng, 2017-07-13:
> access the regular file system from inside RIOT in native.

this would have many (I mean ... lots of) security implications, I guess.

gebart, 2017-07-13:
Well, right now it is possible to access files with regular fopen. You should certainly never run a RIOT application as root.

smlng, 2017-07-13:
oh, wasn't aware of that - and never tried to either ... but sure, using libc calls should be possible.

vincent-d, 2017-07-13:
> so is this a problem/bug and do we need to (or should) address this?

I also think this is an issue.

> access the regular file system from inside RIOT in native.

`real_*` functions have this purpose for board and cpu code I guess.

From an application point of view, a RIOT vfs backend with a local directory mounted inside RIOT would be great IMO.

smlng, 2017-07-13:
> vfs backend with a local directory mounted inside RIOT

👍 certainly!!

closed 2019-09-10 by stale[bot]