RIOT-OS/RIOT #7282: Support for ST X-NUCLEO IKS01A2

smlng, 2017-06-30:
This issue documents the state of driver support for sensor on the [ST X-NUCLEO Shield IKS01A2](http://www.st.com/en/ecosystems/x-nucleo-iks01a2.html), plus to coordinate the efforts and synchronize work on distinct drivers. It is a follow up to #6629 and same hat similar intention as #7152.

Device/Sensor Type | Driver | state
------ | ------ | ----
Humidity and Temperature | HTS221 | 🚧 see #7280  
Accelerometer and Gyroscope | LSM6DSL | ✅ (#6835)
Accelerometer and Magnetometer| LSM303AGR | ❓ maybe adapt LSM303DLHC
Pressure and Temperature | LPS22HB | ❓ maybe adapt LPS331AP

miri64, 2017-06-30:
If they are one shield, unifying them under one driver (in addition to the individual device drivers) like we did with the `io1_xplained` extension board for `samr21-xpro` might make sense.

smlng, 2017-06-30:
mhm, I rather thought about introducing a pseudomodule `iks01a2` which has the respective drivers as a dependency.

miri64, 2017-06-30:
Also thinkable. But don't you have to configure the periphs used by the shield somewhere?

smlng, 2017-06-30:
could be done with a distinct `<DRIVER>_PARAMS_IKS01A2` in `drivers/<driver>/include/<driver>_params.h` and then using `#ifdef MODULE_IKS01A2` I guess? But again: first things first, i.e. getting all drivers ready and afterwards we care about the shield config.

miri64, 2017-06-30:
> But again: first things first, i.e. getting all drivers ready and afterwards we care about the shield config.

Yes, just wanted to bring this up.

smlng, 2017-06-30:
duly noted, thanks for the pointer to `io1_xplained` - wasn't aware of that.

smlng, 2019-08-23:
closing this, its just a tracking issue.

closed 2019-08-23 by smlng