RIOT-OS/RIOT #7287: sys/cbor: strptime implicit declaration

tprrt, 2017-06-30:
The function cbor_deserialize_date_time does not build on ATmega targets because the strptime function is not declared.

cbor.c: In function ‘cbor_deserialize_date_time’:
cbor.c:769:5: error: implicit declaration of function ‘strptime’ [-Werror=implicit-function-declaration]
     if (strptime(buffer, format, val) == 0) {
     ^
cc1: all warnings being treated as errors

miri64, 2017-07-01:
A solution for that could be to provide an implementation (or get it from an LGPLv2.1 source) of that function in `cpu/atmega_common/` and add the function to the `cpu/atmega_common/avr-libc-extra/time.h` header.

gebart, 2017-07-27:
Another solution would be to disable the cbor_ctime module for atmega

miri64, 2017-10-24:
I've collected all `cbor` related issues in #7800. Let's discuss there how to go on with that module.

closed 2017-10-24 by miri64