RIOT-OS/RIOT #7294: RFC: merge candev API into netdev API

miri64, 2017-07-02:
Looking at the newly merged candev API I still don't see, why this has to be its own API apart from netdev:

* The receive mechanism is nearly the same (except that there is no receive function, I'm guessing the frame is passed via the event callback).
* The only difference in the two `send()` is that `can` gives the explicit frame, instead of a vector array with the data
* `init()`, `isr()`, `set()` and `get()` are the same (the latter to just get a new option set)
* `abort()`, `set_filter()` and `remove_filter()` can be implemented using `set()` and corresponding `NETOPT`s

I understand that CAN's communication paradigm is different from how other packet-switched radios work, so there might never be an IP over CAN ([but people tried](https://tools.ietf.org/html/draft-cafi-can-ip-00)), but at least for a programmer it might be nice to not get used to a same but different API.

vincent-d, 2017-07-03:
> Looking at the newly merged candev API I still don't see, why this has to be its own API apart from netdev:

I disagreed with that, but I must say that it could make sense. I still need to think a bit more about it.

> The receive mechanism is nearly the same (except that there is no receive function, I'm guessing the frame is passed via the event callback).

You are guessing right. But I believe the flow could be changed to match netdev's one.

> The only difference in the two send() is that can gives the explicit frame, instead of a vector array with the data

Yes, and IMO this is mandatory to pass the frame explicitely. But it could be encapsulated in an iovec.

> abort(), set_filter() and remove_filter() can be implemented using set() and corresponding NETOPTs

The goal of different functions was to avoid useless complexity and use typed data, but it could be translated to a generic get/set somehow I guess.

It might be worth trying to unify these both interfaces. I'll give a try in the next weeks.

vincent-d, 2017-07-05:
I thought a bit more about it and I need some more feedback.

If we start to merge candev and netdev, do you think it would make sense to also use gnrc_netdev?

Currently the architecture is as follow (`+=====+` are threads):
```
+------------+ +------------+
|  conn raw  | | conn isotp |
+------------+ +------------+

               +============+
               |    isotp   |
               +============+

+---------------------------+
|          raw.h            |
+ - - - - - - - - - - - - - +
|                           |
|      dll/router/pkt       |
|                           |
+ - - - - - - - - - - - - - +
|          dll.h            |
+---------------------------+

+============+ +============+
|   candev   | |   candev   |
+============+ +============+
```

But if we use gnrc_netdev and netapi/netreg, it would lead to something like:
```
+------------+ +------------+
|  conn raw  | | conn isotp |
+------------+ +------------+

               +============+
               |    isotp   |
               +============+

+===========================+
|                           |
|       can "router"        |
|                           |
+===========================+
+---------------------------+
|       gnrc_netapi         |
+---------------------------+

+============+ +============+
| gnrc_netdev| | gnrc_netdev|
|  (can dev) | |  (can dev) |
+============+ +============+
```

There would be an extra thread, which is not optimal IMO. 

Moreover, the can router part, could be compared to gnrc_netreg (even if I'm not sure to completely understand netreg) where the routing is not based only on a demux context but a couple frame id / mask.

miri64, 2017-07-06:
That would be even greater. From what I can see `can_router` indeed looks very similar to `gnrc_netreg`, so an extra thread might not be required.

`gnrc_netreg` is in its core just a per-`nettype` lookup-table (where `nettype` usually identifies a protocol, though there are exceptions, like IPv6-extension headers). This lookup-table stores linked-lists of `netreg_entry` objects, that contain a `nettype`-specific demux context (e.g. ports in UDP or TCP, next-header fields in IPv4/IPv6, etc.; `GNRC_NETREG_DEMUX_CTX_ALL` means any context in any `nettype`) and a target (in default configuration the handling thread's PID). `gnrc_netapi` then can use this information to dispatch command messages by type and demux context. Does that help you understand `gnrc_netreg` better?

vincent-d, 2017-07-07:
Thanks for the explanation.

I think we might be able to do something with some adaptations in `gnrc_netreg` and `gnrc_netdev`.

When receiving, if the `gnrc_netdev` sends the CAN id of the received frame as `demux_ctx`, then `gnrc_netreg` might be able to compare the received id with a CAN filter previously registered.

Updates would be needed on `gnrc_netdev` level, because the:
```
static void _pass_on_packet(gnrc_pktsnip_t *pkt)
{
    /* throw away packet if no one is interested */
    if (!gnrc_netapi_dispatch_receive(pkt->type, GNRC_NETREG_DEMUX_CTX_ALL, pkt)) {
        DEBUG("gnrc_netdev: unable to forward packet of type %i\n", pkt->type);
        gnrc_pktbuf_release(pkt);
        return;
    }
}
```
would need to pass the CAN id as `demux_ctx`.

Then `gnrc_netreg` would need update to not use a 'simple' comparison, but a 'filter-based' (`(demux_ctx & filter->mask) == filter->can_id`).

Further question if this can be done, would be: how do we send 'events' through the stack (for instance, tx complete), and how do we handle power management?

vincent-d, 2017-07-07:
I'm also not sure to understand how multiple interfaces are handled by the upper layers. Because the `gnrc_netreg_entry_t` does not contain any reference to an interface.

haukepetersen, 2017-07-13:
Actually, I don't think using `gnrc_netreg` and `gnrc_netdev` would be the way to go here! They pull in a lot of `gnrc` dependencies, that IMHO don't make sense to have in the CAN stack (and that have proven to be sub-optimal at times).

Merging `candev` into `netdev` is however a very good idea (if I remember correctly this was also already subject of discussion in the original CAN PR). I agree full with @miri64, that the mapping should be quite straight forward. 

What I don't quite understand about the current architecture: in which thread's context is the router code running?

vincent-d, 2017-07-13:
> What I don't quite understand about the current architecture: in which thread's context is the router code running?

It runs from the device thread. From my understanding, it's really an equivalent of `gnrc_netreg`.

This router contains a list of registered filters per device. When a frame is received, the `can_router_dispatch_rx_indic()` function is called (through `can_dll_dispatch_rx_frame()` which first allocate the `pkt`) from the device thread. It goes through the list to find which thread(s) subscribed to that frame and send a msg (either msg or mbox) to each of them.
The 'user' threads must then free the frame when they have finished. Once all threads have freed it, the memory is actually freed.

haukepetersen, 2017-07-13:
oh damn, I forgot that the can stack is using the `gnrc_pktbuf` already. Maybe forget my comment above...

miri64, 2017-07-15:
> I'm also not sure to understand how multiple interfaces are handled by the upper layers. Because the gnrc_netreg_entry_t does not contain any reference to an interface.

The interface identifier in GNRC is the `pid`. As for handling in other layers: GNRC typically replaces the link-layer-specific header (from what I understand this would be `can_dll`?) with a generic header [`gnrc_netif_hdr`](http://doc.riot-os.org/group__net__gnrc__netif__hdr.html) that contains a `if_pid` for later reference. If that isn't possible for some very specific reason, I think it is also possible to just prepend that header in the scope of CAN.

closed 2019-09-10 by stale[bot]