RIOT-OS/RIOT #7301: tests: Delay being "u" instead of "ul" breaks delays on 8-bit/AVR platforms

roberthartung, 2017-07-03:
Running tests on 8-bit platforms such as the AVR sometimes fails, because a delay of 100*1000u results in 100000u which does not fit into 16bit unsigned values on 8-bit/AVR systems. This can be solved by either introduction SLEEP_100US variables for each architecture or simply by changing the delay to 100*1000ul.

miri64, 2017-07-03:
(fixed the markdown because of broken output)

miri64, 2017-07-03:
Can you specify *which* tests are failing?

miri64, 2017-07-03:
Usually the timer works with `uint32_t` which shouldn't have this kind of restriction. That's why I'm asking.

roberthartung, 2017-07-04:
@miri64 Oh right, I didnt say which tests ;) I am using the adxl345 test for example. I changed it from `100 * 1000U` to `100 * 1000UL`. You are right that the `xtimer_usleep` is using a `uint32_t` argument, but I guess that the compiler turns the 100*1000U into an unsigned first which is 16-bit? and that overflows I guess?

miri64, 2017-07-04:
Ah right.... Just saw, that the multipliers in `timex.h` have the same problem :-/.

smlng, 2017-07-06:
I think @roberthartung is right, IIRC we had this issue in other places, too - and `UL` solved it

smlng, 2017-07-06:
I encountered it when working with Arduino (Uno, Duemilanove) based on atmega328p which is 8bit, too

smlng, 2017-07-17:
@dnahm can you verify this on arduino uno?

roberthartung, 2017-11-22:
@dnahm @smlng *ping*

closed 2019-09-10 by stale[bot]