RIOT-OS/RIOT #7304: CC2538 RF driver dislikes fast ACKs

Gr3yh0und, 2017-07-03:
Hey,
I'm using two CC2538DK, one with the gnrc_border_router fw and another with my own. However as soon as I configure both with the AUTOACK and ACK_REQ options I get problems when I am sending packets fast.
E.g. when I send two packets immediately from the client to the router, the router/slip only sends one ACK for the second packet.

Has anyone tested that?

roberthartung, 2017-07-04:
Quick notice: the same happened to me on the AT86RF2XX where I tried to send packages as fast as possible which resulted in missing Interrupts and strange behaviours.

Gr3yh0und, 2017-07-04:
Thanks for the feedback, so it looke like a more general issue. I am currently on the 2017.04 Release (as additional information).

smlng, 2017-07-05:
FYI: the cc2538 does not handle ACKs, they are currently filtered out - cause the device cannot handle them, which has to be done in a software MAC layer. 

In general, you get strange behaviour when sending either to fast, i.e., lots of small packets or large packets that get fragmented by 6LoWPAN for instance.

Gr3yh0und, 2017-07-11:
It looks like the cc2538 is able to do hardware acks - or am I mistaken? I don't have much experience on that field but [Contiki ](https://github.com/contiki-os/contiki/blob/master/cpu/cc2538/dev/cc2538-rf.c#L365)and [OpenThread ](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2538/radio.c#L255) are simply setting the `RFCORE_XREG_FRMCTRL0_AUTOACK` which is according to [here](http://ala.isti.cnr.it/svn/wnlab/MONSTER/it.cnr.isti.monster.firmwares/ZStack%202.6.2%20for%20IAR%208.30/Z-Stack/Components/driverlib/cc2538/inc/hw_rfcore_xreg.h) responsible for sending hw acks. Maybe thats a way how to do it as well in RIOT?

smlng, 2017-07-11:
IIRC that's the case in RIOT as well, if a frame with ACK flag set is received the cc2538 sends an ACK back. The problem is rather the other way around, when the cc2538 sends a frame with ACK flags set and the receiver sends an ACK back, the cc2538 does not have any routines to handle the ACK internally but would pass it up to the (software) driver, that is RIOT. Currently, there is no software MAC for the cc2538 enabled that would handle such ACKs. 

Please also see  #5869 and reference discussion in their for further information on this.

Gr3yh0und, 2017-07-11:
Ah thanks, for the heads up. Now I'm getting it and it makes sense though.

PeterKietzmann, 2019-10-22:
IMO this is still present so I'm reopening it. @jia200x I'm mentioning you just so you're aware of this.

miri64, 2019-10-22:
Given https://github.com/RIOT-OS/RIOT/issues/7304#issuecomment-312794127 this might be related to #11256.

jia200x, 2019-10-23:
> Given #7304 (comment) this might be related to #11256.

IMO that completely explains the problem. The framebuffer gets a lot of IO operations with soft ACKs

roberthartung, 2019-10-23:
@miri64 I would be happy to run some tests on this!

miri64, 2020-07-01:
> @miri64 I would be happy to run some tests on this!

_waits several month with reply_ That would be great!

maribu, 2022-09-16:
ping?

miri64, 2022-09-19:
Could have been solved with the move to 802.15.4 HAL. However, this should be tested. @jia200x maybe?

jia200x, 2022-09-19:
this is definitely not present anymore, as we have performed [severe stress tests](https://github.com/RIOT-OS/RIOT/pull/16946) and never saw an issue like that.
I propose to just close this one.

closed 2022-09-19 by maribu