RIOT-OS/RIOT #7328: Avoid the use of goto

nikosft, 2017-07-06:
Using goto is a terrible programming practice. There might been some benefits for using in the core code (I cannot see any), but for god sake do not use it in the examples. 

The file that caught my attention is RIOT/blob/master/examples/gcoap/gcoap_cli.c which in line 139 has a goto statememt, which can be simply replaced with the code under the label end: (only two lines).

smlng, 2017-07-06:
for the gcoap example you're certainly right. On the other hand there are roughly 30 (something) usages of goto in RIOT overall, considering that this is an (embedded) IoT operating system, that count is rather low. 

So, please feel free to provide a PR fixing the gcoap example, but otherwise I don't see this as an issue (at least for now).

kaspar030, 2017-07-06:
> Using goto is a terrible programming practice.

That is debatable. Using goto for error handling in C is actually considered good practice by many, as it often leads to more readable, less error prone code.

kaspar030, 2017-07-06:
> gcoap_cli.c which in line 139 has a goto statememt, which can be simply replaced with the code under the label end: (only two lines).

You mean, duplicate the printf call?

smlng, 2017-07-06:
> You mean, duplicate the printf call?

I think the code can be easily reworked using `if .. else if ... else` avoiding `goto` without affecting readability - which is not possible in all cases where `goto` is used.

kaspar030, 2017-07-06:
>> You mean, duplicate the printf call?
> I think the code can be easily reworked using ...

Sure. I wanted to point out that just copying the two lines is not gonna cut it.

nikosft, 2017-07-06:
It is not about reducing the line of codes. It is about readability and good code practices. When goto is used instead of if...else.... then "it is considered harmful"

closed 2017-07-06 by nikosft