RIOT-OS/RIOT #7330: Ethos does not compile

nikosft, 2017-07-07:
When USEMODULE += ethos is included in an app's Makefile, compiling the app fails with error "comparison between signed and unsigned integer expressions" in RIOT/drivers/ethos/ethos.c: lines 303 and 311.

To solve this problem line 303 should be replaced from
``if (len < (int)dev->last_framesize)``
 to
``if (len < dev->last_framesize)``

and line 311 from
``if ((tsrb_get(&dev->inbuf, buf, len) != len))``
to
``if ((tsrb_get(&dev->inbuf, buf, len) != (int)len))``

Moreover, Readme.md in RIOT/dist/tools/ethos should be updated to include that app Makefile should also include USEMODULE += uart_stdio

smlng, 2017-07-10:
thanks for reporting, are you willing to provide a fixing PR?

miri64, 2017-10-25:
I will look into that after the release.

kYc0o, 2018-01-16:
I tried to reproduce this error but it works ok so far. Though it's true that `USEMODULE += uart_stdio` is necessary (a dependency).

I did it by typing:

```console
BOARD=samr21-xpro USEMODULE+=ethos USEMODULE+=uart_stdio make
```

on the default example.

@nikosft can you confirm if your issue persists?

kYc0o, 2018-01-30:
I retested on current master and found that it compiles right with the configuration provided on the README. I'll close it for now, reopen if you find the error again.

closed 2018-01-30 by kYc0o