RIOT-OS/RIOT #7332: Real Time Timer that supports thread sleep/yield

photonthunder, 2017-07-07:
In my application I have several threads that are asleep most of the time but wakeup at periodic times to do some work.  I use the RTT alarm before sleeping the thread:

```
rtt_set_alarm(sleepDuration + startTime, rttCallback, NULL);
thread_sleep();
```

And then wake with the callback:

```
void rttCallback(void *arg) {
    thread_wakeup(MainPid);
}
```

What I would like is a method for supporting waking up using the RTT (or other timer) from multiple threads.  I had hoped initially XTIMER would do the job but as is well documented it doesn't allow sleep.

Does this feature exist and I am just missing it?

gebart, 2017-07-09:
There is no inherent limitation in xtimer itself that prohibits low power operation, the problem is that the xtimer configurations for most platforms are using the hardware timers which stop counting in sleep modes. On most CPUs there is at least one timer which keeps running in low power mode, for example for Kinetis it's the LPTMR module, for STM32 there's one as well. The downside is that the low power timers are usually limited in other features, such as lower accuracy or lower frequency. xtimer is designed to allow setting timers on 1 µs precision, which requires a 1 MHz hardware timer, but you can still use xtimer with reduced accuracy if the hardware timer is slower than 1 MHz.

If you want to try with a low power timer you need to update the board.h for your board and add or change the definition of XTIMER_DEV and all related constants. See an example of two different configurations for a Kinetis CPU (mulle) here: https://github.com/RIOT-OS/RIOT/blob/master/boards/mulle/include/board.h#L42:L69

Worth mentioning is that @haukepetersen and @kaspar030 have said in other threads on this tracker that they are working on plans for an improved xtimer which would use two hardware timers together to allow both high accuracy when needed, and low power operation when the accuracy is not needed. This seem to still be a bit into the future, as there is not yet any concrete proposal or any public code for this improved system.

photonthunder, 2017-07-10:
@gebart thanks for the information and link.  One approach I have seen used elsewhere is switch to the your low power clock (like your 32 kHz external crystal) before sleep at set the expected sleep time.  Then when you wake your update your xtimer clock based on how long you sleep.  The drawback is that you do introduce some jitter/noise into your timer clock.

closed 2019-09-10 by stale[bot]