RIOT-OS/RIOT #7334: samd21: DFLL configuration

Hyungsin, 2017-07-08:
DFLL setting at https://github.com/RIOT-OS/RIOT/blob/master/cpu/samd21/cpu.c#L163 
may flush the previous setting at https://github.com/RIOT-OS/RIOT/blob/master/cpu/samd21/cpu.c#L158.

How about "SYSCTRL->DFLLCTRL.bit.ENABLE = 1" instead?

smlng, 2017-07-13:
the questions is, if it is a bug or whether this might be intentional? Did you observe any problems? In general you right, codewise current implementation this will flush/overwrite the DFLL register setting - but again, might be on purpose.

photonthunder, 2017-07-13:
It is a bug, the |= got dropped somehow.  I will get a PR going to add it back in.  The change means that you are running in open loop mode, meaning you are not taking advantage of the external crystal to calibrate the DFLL.

smlng, 2017-07-13:
okay, where was it dropped - was there a PR introducing the regression or already when the CPU got ported for RIOT? Otherwise I'd rather got for your suggested:

```
SYSCTRL->DFLLCTRL.bit.ENABLE = 1;
```

instead of using '|=', the above is consistent with the rest of the code and more explicit.

photonthunder, 2017-07-13:
PR #7130 added the option of using DFLL and in my initial code the |= existed, but at some point it was dropped.  I will push a PR with the one line fix.

Hyungsin, 2017-07-13:
@photonthunder, Cool. Go for it! :)

photonthunder, 2017-07-13:
PR #7359 created, thanks for catching this bug.

Hyungsin, 2017-07-13:
This is closed thanks to PR #7359.

closed 2017-07-13 by Hyungsin