RIOT-OS/RIOT #7347: xtimer_usleep stuck for small values on nucleo-l1

MichelRottleuthner, 2017-07-10:
On `nucleo-l1` (L152RE) `xtimer_usleep` never returns if you hand it values <=11.

minimum example to reproduce the error:
```c
#include <stdio.h>
#include "xtimer.h"

int main(void)
{
    for (int i = 100; i > 0; i--){
        printf("going to sleep %d usecs...\n", i);
        xtimer_usleep(i);
    }
    return 0;
}
```
For me this always leads to the application getting stuck after printing the following line: 
```
going to sleep 11 usecs...
```
In xtimer.c the callback [`_callback_unlock_mutex`](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L42) that should be fired after [`_xtimer_set64(&timer, offset, long_offset)`](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L64) is never called.

I also did a quick test on `nucleo-l476` where the faulty behaviour could not be observed.

smlng, 2017-07-11:
mhm, I cannot reproduce with `nucleo-f411` either. However, it might have something to do with the xtimer config in `board.h`. For instance the `nucleo-l476` does not have an explicit xtimer config (see [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-l476/include/board.h) ), hence default values will be used. Try to delete the xtimer config [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-l1/include/board.h#L34), so defaults kick in for the `l1`, too.

smlng, 2017-07-11:
well you know, I can reproduce on f411, now 😕 but it gets stuck at `going to sleep 7 usecs...`

smlng, 2017-07-11:
guess we should try you test on every board ...

smlng, 2017-07-11:
mhm, changing `XTIMER_BACKOFF` from `5` to `40` (RIOTs default) fixes the issue. Seems like there is some kind of race condition, when the timer is actually set or it simply spins. Maybe the backoff values are chosen by dice for many platforms without actually verifying them. 

Can you please provide a PR with your small test to have in the repo, something like `tests/xtimer_usleep_tiny`?

smlng, 2017-07-11:
I just tested with PhyNode (`pba-d-01-kw2x`) which has no board config either, and hence uses default xtimer config, i.e. with `XTIMER_BACKOFF (30)` - which works. However, explicitly settings this value to anything `< 22` makes your test hang at 23us - similarly as I reported above the `nucleo-f411` got stuck at 7us with set `XTIMER_BACKOFF (5)` which makes a diff of 2us between backoff time and when it gets stuck. Looks like a race condition when `XTIMER_BACKOFF` is very close to the desired sleep interval (which decides over setting a timer or spin), because there is also a slight processing overhead between setting and actually doing the sleep - maybe those 2us. 

As already said above, we should check all boards, especially those with an explicit xtimer config in their `board.h` (because default values seems to be rather safe), if they succeed in setting very low sleep intervals

MichelRottleuthner, 2017-07-11:
Thanks for testing and providing additional info! I will create a PR for the test as you suggested.

MichelRottleuthner, 2017-07-11:
>Try to delete the xtimer config here, so defaults kick in for the l1, too.

Just tested that to be sure and as expected this fixes the Problem for the `nucleo-l1`.

gebart, 2017-07-11:
I had a similar problem on the frdm-kw41z board, but the timer used had a wrap around of 2 seconds, so it only hung for 2 secs. I fixed it by increasing XTIMER_BACKOFF and related constants by a small amount. I think you'll have to do some trial and error experiments to find the correct values.

MichelRottleuthner, 2017-07-11:
Only modifying XTIMER_BACKOFF the smallest value that fixes the problem for me is 10. But I wonder if there are other factors like compiler optimization etc. that may cause different behaviour on this. Is trial and error the current best practice to determine the right value for XTIMER_BACKOFF?

smlng, 2017-07-11:
Or rephrase the question to "What is the impact of using xtimer default values instead of board specific optimizations?" that is `XTIMER_BACKOFF (40)`, for instance. Certainly, if xtimer backoff is to high it will often spin instead of setting an actual timer, which might have performance impacts.

Further, we really need to look into the other config params, e.g., what about the timer frequency, default is 1MHz but on (some, 8Bit) arduino its 250kHz.

smlng, 2018-01-31:
any news, how to proceed here -> fix would be to adapt XTIMER_BACKOFF and stuff in board.h right?

Marc-Aurele, 2018-01-31:
Hi,

In my case, i faced two kinds of issues when i was calling xtimer_usleep
with small values :
 - my task was stuck in sleep state and was never waking up after
calling xtimer_usleep  => and indeed, incrementing XTIMER_BACKOFF seems to
have solved the issue
 - my task was stuck in running state in a "while" loop => didn't find a
solution yet, i have incremented my AHB bus speed but i rely on my watchdog
to get out of this state :s

Regards,

Aurélien

MichelRottleuthner, 2018-03-20:
As #8763 only speaks of l152, and there were some other boards mentioned above: is this really fixed for "all" boards?

kYc0o, 2018-03-20:
No, it only fixes for L1 family. However I had no issues with L151, but #8763 doesn't affect the output so it's ok. I have errors in F446 so I assume the F4 family "needs" a backoff of 7 to work properly.

aabadie, 2018-03-20:
Reopening the issue since other boards are affected (nucleo-f411, pba-d-01-kw2x,  frdm-kw41z).

Marc-Aurele, 2018-03-20:
in my case, it was on a nucleo-l073 and on my custom boards thats embeds the same cpu. Since i have incremented the AHB bus speed and the XTIMER_BACKOFF the issue doesn't occur anymore ... but i'm consuming a little bit more at wake up now :(

miri64, 2020-07-01:
@MichelRottleuthner is this still a problem? @aabadie can you maybe test?

pokgak, 2020-07-01:
I'm currently writing tests that automate testing this for the HIL. I'll run it and report back later.

pokgak, 2020-07-01:
I used the test code given above with a `USEMODULE += xtimer` in the Makefile.

| Board | Result |
|----------|------------|
| saml10-xpro | FAIL (stuck after sleep 38 usecs) |
| samr21-xpro | PASS |
| nucleo-l073rz | PASS |
| nucleo-f411re | PASS |
| remote-revb | PASS |
| arduino-mega2560 | PASS |
| nucleo-f103rb | PASS |
| frdm-kw41z | PASS |
| nucleo-f091rc | PASS |
| esp32-wroom-32 | PASS |
| arduino-due | PASS |

maribu, 2022-09-16:
With xtimer on its way out and ztimer being used instead now every, I'd say this is a wontfix.

If anyone disagrees, please reopen.

PR #8763 (2018-03-20): boards/nucleo-l152: increase xtimer backoff

### Contribution description

This PR just increases the xtimer backoff value for the STM32L1 based boards. It has been tested on nucleo-l152re board and fixes 2 stucked xtimer based test applications: `xtimer_usleep_short` and `xtimer_periodic_wakeup`.

Other xtimer test applications are still passing.

### Issues/PRs references

Fixes #7347
closed 2022-09-16 by maribu