RIOT-OS/RIOT #7384: RFC: Expose OpenThread API via ot_exec_commands

jia200x, 2017-07-19:
OpenThread has API functions to interact with the OT core (functions with ot prefix like otUdpReceive(), otUdpNewMessage(), available in the original OpenThread docs). With the current port of OT in the master branch, it's not possible to directly call these API functions due to concurrency problems, so there's an [ot_call_command](https://github.com/RIOT-OS/RIOT/blob/master/pkg/openthread/contrib/netdev/openthread_netdev.c#L54) function which ensures all otXXX functions are called in the same thread running the OpenThread core.

For instance,  the openthread test [sets the PAN ID with this method](https://github.com/RIOT-OS/RIOT/blob/master/tests/openthread/main.c#L29), and this will end up executing [this function](https://github.com/RIOT-OS/RIOT/blob/master/pkg/openthread/contrib/platform_functions_wrapper.c#L219) which calls otXXX functions safely.

The problem I see with this method is the user is not able to directly execute otXXX functions and only has [a predefined set of commands](https://github.com/RIOT-OS/RIOT/blob/1a2097e97e90daa4479d5c9ce4dcecd3f57d31dc/pkg/openthread/contrib/platform_functions_wrapper.c#L56). At some point in the development of the port there was an ot_exec_job that did the same trick, but allowed to define otXXX snippets and run them within the same file.

So, instead of doing this:

`ot_call_command("panid", ... ... );`

One could do this:

```
OT_COMMAND ot_panid(otInstance* ot_instance, void *context) {
    /* Safely calling an OpenThread API function */
    otLinkSetPanID(...);
    ...
}

`ot_exec_command(ot_panid, the_context);`
```

There's an outdated branch that shows what I mean [here](https://github.com/Inria-Chile/RIOT/blob/b13475a1e1b4d258f311eaee222206e388fa47bd/tests/openthread_test/main.c#L195)

Probably looks not-so-clean but won't require to modify the pkg files each time a new OpenThread feature is added and seems to be more flexible to me.

Any comments? ;) 

@biboc @aabadie

biboc, 2017-07-24:
How do you 
> /* Safely calling an OpenThread API function */
?

biboc, 2017-07-24:
Ah! I understand now! That's great and this is why as an example you showed how to call a function inside OpenThread thread via ot_exec_job so all functions have made in platform_functions_wrapper are example of function you want user to create, true?

So we could put all functions from platform_functions_wrapper in a separate file in tests/openthread to show some examples on how to call OT functions?

Let's merge  #7149 before and we'll do that after

jia200x, 2017-07-24:
>  so all functions have made in platform_functions_wrapper are example of function you want user to create, true?

Exactly ;-)

> So we could put all functions from platform_functions_wrapper in a separate file in tests/openthread to show some examples on how to call OT functions?

I think it's a really good idea. Also, each module could have it's own set of OT_JOBS (e.g sock).

> Let's merge #7149 before and we'll do that after

I agree!

jia200x, 2019-08-12:
I think it could make more sense for us to use GNRC on top of OpenThread's MLE/MAC layer. That's the approach other OS have followed.
So, we could just reuse sock/gcoap without having to write new wrappers.
In fact, OpenThread provide hooks for stacking an IP layer on top.

I will then close this issue.

closed 2019-08-12 by jia200x