RIOT-OS/RIOT #7411: Error (WERROR=1) while compiling the CBOR module

kYc0o, 2017-07-25:
While testing some stuff in the mulle platform I notice that the unittests test doesn't compile anymore. This is due to the CBOR module which throws the following error (warning):

```console
/Users/facosta/git/RIOT2/RIOT/sys/cbor/cbor.c: In function 'cbor_deserialize_date_time':
/Users/facosta/git/RIOT2/RIOT/sys/cbor/cbor.c:770:9: error: implicit declaration of function 'strptime' [-Werror=implicit-function-declaration]
     if (strptime(buffer, format, val) == 0) {
         ^~~~~~~~
cc1: all warnings being treated as errors
```

I tried several methods to make it disappear without success.

Besides, the CBOR test (not unittest) has now a stack overflow probably due to #7230 but I still need to verify.

miri64, 2017-07-26:
I tried

```bash
WERROR=1 BOARD=mulle make -C tests/unittests/ -B all
```

and did not encounter any errors. Which toolchain are you using?

miri64, 2017-07-26:
Alternatively, use `git bisect` to determine the last working commit.

kYc0o, 2017-07-26:
For the cbor test I get this:

```console
snake:RIOT facosta$ cd tests/cbor/
snake:cbor facosta$ BOARD=mulle make all
Building application "tests_cbor" for "mulle" with MCU "k60".

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/boards/mulle
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/k60
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/cortexm_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/cortexm_common/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/k60/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/kinetis_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/cpu/kinetis_common/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers/mtd
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers/mtd_spi_nor
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers/nvram
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers/nvram_spi
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/drivers/periph_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/sys
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/sys/auto_init
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/sys/cbor
/Users/facosta/git/RIOT2/RIOT/sys/cbor/cbor.c: In function 'cbor_deserialize_date_time':
/Users/facosta/git/RIOT2/RIOT/sys/cbor/cbor.c:769:9: error: implicit declaration of function 'strptime' [-Werror=implicit-function-declaration]
     if (strptime(buffer, format, val) == 0) {
         ^~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [/Users/facosta/git/RIOT2/RIOT/tests/cbor/bin/mulle/cbor/cbor.o] Error 1
make[2]: *** [ALL--/Users/facosta/git/RIOT2/RIOT/sys/cbor] Error 2
make[1]: *** [ALL--/Users/facosta/git/RIOT2/RIOT/sys] Error 2
make: *** [link] Error 2
snake:cbor facosta$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

For the unittests I get exactly the same.

I also got the same error with gcc 5.2 (I updated it because I thought that was the problem).

miri64, 2017-07-26:
Mh I don't get that error with `tests/cbor` either.

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

kYc0o, 2017-07-26:
Should we blame my libc version then? I thought it came with the toolchain...

miri64, 2017-07-26:
I guess so.

gebart, 2017-07-26:
I saw something similar the other day in another situation. I can try it with my toolchain when I am back at my PC.

gebart, 2017-07-27:
cbor.c is missing a block
```c
#ifdef MODULE_CBOR_CTIME
#include <time.h>
#endif
```

Not sure that is the proper fix though, there might be a missing `#define _DEFAULT_SOURCE` as well

miri64, 2017-07-27:
I just noticed, that there is another issue (#7287) that is referring to this error.

miri64, 2017-10-24:
I've collected all `cbor` related issues in #7800. Let's discuss there how to go on with that module.

PR #7783 (2017-10-24): sys/cbor: fix compilation with newlib
Fixes #7411 and maybe #7287.

In some arm-none-eabi-gcc libc `strptime` in`<time.h>` is guarded by `__GNU_VISIBLE` define, which is defined while setting the define `_GNU_SOURCE`, otherwise `strptime` is not defined and thus any usage of it fails to compile.

This is the case in cbor, which doesn't include `<time.h>` explicitly, as well as setting the define `_GNU_SOURCE` is missing.

This PR fixes that.
closed 2017-10-24 by miri64