RIOT-OS/RIOT #7418: Buffered printf()

travisgriggs, 2017-07-26:
I would like our printfs to run in the background, or rather the feeding of them out the serial port. Is there a general recipe for how to accomplish this in the RIOT framework?

gebart, 2017-07-27:
If you still want the messages to be printed on the UART you will need to modify the UART driver for your CPU to use interrupts or DMA and it will probably need an extra internal buffer to keep while the UART is working.

travisgriggs, 2017-08-03:
Thanks @gebart.

kaspar030, 2017-08-04:
A simple way would be to create a thread reading from an isrpipe writing to uart, then modifying the newlib write() implementation to write to that isrpipe. The isrpipe would double as buffer. The cost would be an additional thread stack and the buffer, and still, as uart writes are synchronous, after the buffer is full, printf's would block. Might still be useful for debugging.

travisgriggs, 2017-08-07:
Regarding "... then modifying the newlib write() implementation to write to..." I've never done that before, can you give a pointer on how one does that? From what little I can glean from Google, newlib is the version of libc used for RIOT?

gebart, 2017-08-08:
@travisgriggs the interface between newlib and RIOT are called syscalls, and are implemented in https://github.com/RIOT-OS/RIOT/blob/master/sys/newlib_syscalls_default/syscalls.c (Also note that there are two implementations of the file-related syscalls, one with the vfs module and one without it, which one you are using depends on whether you have USEMODULE+=vfs in your application, or not)

closed 2019-09-10 by stale[bot]