RIOT-OS/RIOT #7430: Explain how _snd_rcv works.

holasm, 2017-07-31:
I am trying to understand the flow of packets. 
I came across the functions *_snd_rcv * in *sys/net/gnrc/netapi/gnrc_netapi.c*.

Can anyone explain how the function works.
Specially what is `sendto->target.pid`

thanks

miri64, 2017-07-31:
Hello @holasm, welcome to RIOT.

First of all, you have to understand, that the GNRC network stack has a multithreaded design and consists of one thread per each module (that's why they are addressed by PID). Thus, the difference between receive and send from a packet's perspective is just, if it moves up (receive) or down (send) the network stack. This is why both NETAPI commands / messages can be dispatched in the same function.

```C
    msg_t msg;
    /* set the outgoing message's fields */
    msg.type = type;
    msg.content.ptr = (void *)pkt;
```

This lines prepares an [IPC message](http://doc.riot-os.org/structmsg__t.html). `type` can either be `GNRC_NETAPI_MSG_TYPE_RCV` or `GNRC_NETAPI_MSG_TYPE_SND`. The remaining lines then send the message then [in a non-blocking manner](http://doc.riot-os.org/group__core__msg.html#ga99264bd4f263fa2b478308c1f7983c10) to the target module (i.e. they are put in the target's thread message queue).

```C
    /* send message */
    int ret = msg_try_send(&msg, pid);
    if (ret < 1) {
        DEBUG("gnrc_netapi: dropped message to %" PRIkernel_pid " (%s)\n", pid,
              (ret == 0) ? "receiver queue is full" : "invalid receiver");
    }
    return ret;
```

If you want to know how the IPC message is handled then by the receiving thread, have a look at [gnrc_udp.c](https://github.com/RIOT-OS/RIOT/blob/3686b2cba531124b085c9277dc9dfd03fcff0f7e/sys/net/gnrc/transport_layer/udp/gnrc_udp.c#L220-L259).

Did that answer your question?

holasm, 2017-08-02:
So the while(1) loop keeps spinning for new message to be available at `&msg` and it sens or receives accordingly. Could me tell me where the `sendto->target.pid` is assigned.

miri64, 2017-08-02:
A callee of [`gnrc_netreg_register()`](http://doc.riot-os.org/group__net__gnrc__netreg.html#gafb6a9d0b840aae07c3c1617932835ef1) is supposed to set it, before calling that function. In most cases this is done by using the [`gnrc_netreg_entry_init_pid()`](http://doc.riot-os.org/group__net__gnrc__netreg.html#ga806d3268f56159bb73fcbfcc92c5cd2f) helper function or static [`GNRC_NETREG_ENTRY_INIT_PID()`](http://doc.riot-os.org/group__net__gnrc__netreg.html#gaed56e891d9a47cba3dcf80f927623137) helper macro.

miri64, 2017-08-02:
(with that said, I see that the doc is somewhat outdated at this point. Will fix)

miri64, 2017-08-02:
For `sock_udp` e.g. however, this is using `sendto->target.mbox`, by the way.

miri64, 2017-08-02:
> (with that said, I see that the doc is somewhat outdated at this point. Will fix)

See https://github.com/RIOT-OS/RIOT/pull/7438

holasm, 2017-08-04:
Thanks @miri64.

closed 2017-08-04 by holasm