RIOT-OS/RIOT #7431: Order of FEATURE_PROVIDED and Makefile.features

roberthartung, 2017-07-31:
The `sam3` cpu is not declaring `FEATURES_PROVIDED += periph_pm`, but `cc2538` does. However, both depend on `cpu/cortexm_common` which also defines periph_pm as a provided feature.

Unfortunately when I build `examples/hello-world/` the order of modules is a little bit different:

```boards/cc2538dk -> core -> cpu/cc2538 -> cpu/2538/periph -> cpu/cortexm_common -> cpu/cortexm_common/periph```
versus
```boards/arduino-due -> core -> cpu/sam3 -> cpu/cortexm_common -> cpu/cortexm_common/periph -> cpu/sam3/periph```

This for some reason leads to a problem with `cpu/cortexm_common/periph/pm.c` because `FEATURE_PERIPH_PM` is defined for the cc2538dk but not for the sam3.

I know that there are both problems with the build system and the linker and I am not sure what to do about this exactly. Basically I just want to keep track of it for now and see if I find a solution or maybe someone can help me on this.

roberthartung, 2017-08-04:
Makefile.features will be automatically included via `makefiles/feaures.inc.mk`. However, if a CPU uses a common CPU that provides features, no Makefile.features can be provided. The actual CPU always has to define FEATURES_PROVIDED variable.

A special case is the following:

The arduino-due board with the sam3 CPU is using the cortexm_common. The cortexm_common CPU has a pm.c and should therefore provide FEATURES_PROVIDED for all of its "children".
A problem arises with the `nrf51` which uses both the cortexm_common AND the nrf5x_common. Both of which partially provide functions for the pm module.

I will try to work on a solution to remove the weak definition and to allow definition of pm functions in different levels (cpu, common, or the default implementation!)

roberthartung, 2017-08-04:
@kaspar030 Maybe your solution for submodules for periph modules might solve this whatsoever!

kaspar030, 2017-08-04:
> @kaspar030 Maybe your solution for submodules for periph modules might solve this whatsoever!

You mean like in #7441?

roberthartung, 2017-08-04:
Oh, I wasn't aware of that one, I will try that one first. I was more referring to #7241

roberthartung, 2017-08-04:
@kaspar030 Oh I see, it's brand new :-D I am currently working on the same issue ... sort of, but with a slightly different approach (just removing weak for now and try to use IFDEF/IFNDEF, let's see how messy it gets!)

roberthartung, 2017-08-04:
See also: https://github.com/roberthartung/RIOT/commits/hartung/pm/remove-weak

miri64, 2017-10-25:
I think that was fixed with the pm fallback solution, right? If not re-open, please.

closed 2017-10-25 by miri64