RIOT-OS/RIOT #7446: RFC: Circular log buffer

bergzand, 2017-08-06:
While testing something out, I got a bit annoyed by `uhcpc` spewing log messages directly in my serial console 😞.
I was thinking that on nodes with a bit of ram to spare, a circular log buffer would be a valuable addition. a KB of ram dedicated to log messages, set up as a circular log buffer. This could then be read out by a shell command or even over CoAP in environments where there is no shell access.

gebart, 2017-08-06:
Kind of like syslog with logrotate on Linux? I like this idea, especially being blue to read the log through coap or even through the debugger when trying to fix UART issues :+1:

bergzand, 2017-08-08:
Yes, the linux kernel log (dmesg) might be the better analogy, but the idea is clear.

I also think that if this project is going to support [slipmux](https://datatracker.ietf.org/doc/html/draft-bormann-t2trg-slipmux), this might be the way to do the message logging.

closed 2019-09-10 by stale[bot]