RIOT-OS/RIOT #7459: What is the map of the pins from ENC28J60 to cc2538 open mote?

holasm, 2017-08-07:


smlng, 2017-08-30:
@holasm the ENC28J60 has an SPI interface, try the following:

- connect SI of ENC28J60 to MOSI pin (DIO6) of the OpenMote
- connect SO of ENC28J60 to MISO pin (DIO5)
- connect SCK of ENC28J60 to SCK (DIO4)
- connect CS of ENC28J60 to CS (DIO7)

problem is I haven't found proper documentation of the pinout map for the open mote (yet) so I'm unsure how to connect and configure RESET and INT pins right now. Please have a look at the board config [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/openmote-cc2538/include/periph_conf.h#L144) and the driver tests [Makefile](https://github.com/RIOT-OS/RIOT/blob/master/tests/driver_enc28j60/Makefile#L31), for the latter you'll have to adapt the pin config, as in #7458.

holasm, 2017-09-09:
@smlng Thanks for your response.
According to openmote  [periph_conf.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/openmote-cc2538/include/periph_conf.h#L140) I have set the SPI interface. 
I set the **INT pin to DIO0** according to [this](https://sourceforge.net/p/contiki/mailman/message/33525505/)

For the RESET pin I can't find any doc.

maitysubhasis, 2017-09-12:
@smlng I have figured out the pin mapping with some help of my friend.
[image]

smlng, 2017-09-28:
@maitysubhasis nice and thanks for sharing this here! So is this issue resolved and can be closed?

closed 2018-04-24 by maitysubhasis