RIOT-OS/RIOT #7494: Doxygen: Search box missing in mobile browsers/small screens

gebart, 2017-08-22:
I can't get the search box found in the top right corner of the Doxygen documentation to show up if I open http://riot-os.org/api/ on my phone, even when checking "Request desktop site" in the site settings.
It seems like the cause may be that the screen is too narrow to fit everything on one line. The same symptom shows on the computer if I shrink my web browser window to approximately half the screen width. 

Browser: Chrome 59.0.3071.125, Android 7.0.0

miri64, 2017-08-22:
It's not a bug, it's a feature. The search results (they are created by doxygen-imported javascript) look insufferably unresponsive and not really suitable for mobile (believe me it isn't very user friendly), so I decided to remove it altogether, until doxygen gets a more responsive design by default.

gebart, 2017-08-22:
It is incredibly annoying to not be able to search for something on the mobile, especially when looking for the docs on one particular function without knowing under which module it is located. The only other option is to use a custom Google query to find the correct subpage for the function or typedef I am looking for, which takes a lot more effort than waiting for a slow javascript for a few seconds.

miri64, 2017-08-22:
I know... but until it is fixed in doxygen, there is no way to do it otherwise :-(.

miri64, 2020-07-01:
@kfessel maybe, while fixing #14412 maybe you can have a look into this as well?

kfessel, 2020-07-01:
@miri64 i had a look (firefox in smartphone emulation): the base of the problem is in the doxygen generated search.js it is calculating an offset from the searchbox position to the left  -> the results are displayed out of screen and there for not reachable.

this might be fixable with some javascript in the header html that basicaly removes the offset after doxygen added it

i am not sure about the unresponsiveness emulation is emulation, i need to test with my smart phone

kfessel, 2020-07-02:
https://kfessel.github.io/RIOT/html/ << please check there

miri64, 2020-07-02:
Currenlty it looks like this, so still needs some pixel-schubsing as one might say in German :-P

[image]

kfessel, 2020-07-02:
@miri64 are there still performance issues for you?

your first commet:
> ...The search results (they are created by doxygen-imported javascript) look insufferably unresponsive..

kfessel, 2020-07-02:
for the pixelschubsing the vertical position (the original (if you search on desktop there is also half of the line missing)) 
I can move it down and make the frame a bit slimmer

miri64, 2020-07-02:
> @miri64 are there still performance issues for you?

No, this was also never about performance. With responsiveness I meant it in the bootstrap sense, so that for small screens the width is automatically adapted ;-).

miri64, 2020-07-02:
This does not seem to be an issue though, so :+1:

kfessel, 2020-07-02:
please check again
(force reload)

miri64, 2020-07-02:
see PR :-)

PR #14412 (2020-07-04): doc/search: Patch search to display results bigger if hitting return and work on mobil
### Contribution description

make the search work on Mobil #7494 

make search not reload on hitting return, but display the results in content-part of the window

### Testing procedure

I used Firefox inspector to test

### Issues/PRs references

Fixes #14409 doc/search: hitting return while searching on api.riot-os.org reloads current page #14409
Fixes #7494

This is in conflict with  #14411 only one should be applied, I prefer this one
closed 2020-07-04 by miri64