RIOT-OS/RIOT #7503: gnrc: is it possible to pass flags/extra info between layers?

gebart, 2017-08-23:
Is it possible to pass extra information about a packet from the 6lowpan layer to the link layer for setting link layer protocol bits depending on the content? 

For radio duty cycling MAC layers it is useful to be able to pass information from the higher layers (e.g 6lowpan) to the link layer whether the packet is part of a stream or burst of multiple packets, or if it is the last one. For example, if the 6lo fragmentation module could tell the mac layer that the packet is a fragment, and the 802.15.4 module can set the fcf frame pending bit to tell the receiver to remain on for a short while after reception to receive the rest of the fragments.

miri64, 2017-08-23:
Most (if not all) of this information is in the [6LoWPAN frame](https://tools.ietf.org/html/rfc4944#section-5.3) and need not be specially encoded by some GNRC mechanism.

* Packet is a fragment? Yes, if packet start with bits 11000 or 11100
* Packet is part of a specific collection of fragments (is that what you mean with burst or stream?): Packets share the same datagram tag
* Packet is the last one? This might be a little trickier, since 6LoWPAN really doesn't have a concept of "last packet", but it is encoded in the fragment's length, the datagram length field and the datagram offset field.

If you meant more with streams or bursts (e.g. TCP streams), than this is beyond 6LoWPAN, but some clever setting of the Flow Label in IPv6 might solve that (not sure if this is the intended use case for those, though, if not this information could go into `gnrc_netif_hdr_t`).

miri64, 2017-08-23:
> * Packet is a fragment? Yes, if packet starts with bits 11000 or 11100

Oh, you might need to parse the packet in case [Mesh Headers precede the fragmentation header](https://tools.ietf.org/html/rfc4944#section-5)

gebart, 2017-08-23:
I was hoping that the link layer would not have to know the format of 6lowpan frames, to keep the mac layer independent of what is running on top of it.
It feels kind of redundant to have one thread encode the network header with all the compression magic and extra headers, and then have the next layer decompress it and parse it all, just to set a single bit in the link layer header. Wasteful of resources (cycles and power) and increases latency. 
Are there any protocol independent flags or anything that can be used to pass this kind of info from network layer to link layer?

miri64, 2017-08-23:
There are the [flags in `gnrc_netif_hdr_t`](http://doc.riot-os.org/structgnrc__netif__hdr__t.html#abbf718ba7223770b72eb8fbd39dd85f6) that can be used for that. But there is also some work regarding this for the UDP checksum compression in Hamburg (sorry forgot who is doing the work, @PeterKietzmann can you say?). So there needs to be some consolidation done before randomly choosing flags ;-).

gebart, 2017-08-25:
I will try to do a proof of concept with netif flags

miri64, 2019-08-10:
> @gebart, can this PR be closed?

I guess so.

closed 2019-08-10 by miri64