RIOT-OS/RIOT #7513: core/debug.h: stack size check for DEBUG_PRINT outdated?

haukepetersen, 2017-08-25:
The current check for stacksize in the `DEBUG_PRINT` macro in `debug.h` is mostly rather annoying. The `Cannot debug ...` message pops up on most platforms, when trying to print DEBUG messages in interrupt context. Potentially this is not needed anymore (since we are using the newlib nano specs)?! 

Also, the fallback to outputting a fairly long string using `puts` instead seems quite questionable to me. So I would actually tend to remove the check all together and instead add a clear comment, that one needs to be very careful when using the `DEBUG` macro in interrupt contexts.

gebart, 2017-08-25:
I would not mind getting rid of this check. I have seen this as well.

The memory usage depends on the format string, so it is hard to calculate a suitable limit. The nano printf still uses a lot of stack space because of a very deep call chain until it reaches write(). That said, I still think the check might be a bit on the conservative side regarding the memory limit.

gebart, 2018-11-22:
A related problem I have noticed on Cortex-M is that the stack check from ISR mode is actually checking the size of the stack of whatever thread was running before the interrupt, so if the system is idle before the interrupt, the check is made against the idle stack, which is typically tiny. So only increasing the ISR stack size does make the warning go away unless you also increase the idle stack size (which will be unused since the stack used when calling printf from an ISR is the isr stack).

maribu, 2022-09-19:
The check is still needed even with newlib-nano, but we do have `DEBUG_PUTS()` as a better alternative to just using `puts()` for debugging.

closed 2022-09-19 by maribu