RIOT-OS/RIOT #7543: core: thread_flags: THREAD_FLAG_MUTEX_UNLOCKED, THREAD_FLAG_TIMEOUT are not used

gebart, 2017-08-30:
Follow up to #7533. I found that there are two other predefined constants which are not used in the tree.

- `THREAD_FLAG_MUTEX_UNLOCKED`
- `THREAD_FLAG_TIMEOUT`

I think the constants may be intended for `xtimer_msg_timeout`, which mentions core_thread_flags in the documentation, but does not in fact use thread flags for the implementation.

kaspar030, 2017-09-01:
#7558 implements a simple timeout function using THREAD_FLAG_TIMEOUT, which I think will be used in many timeout variants of existing functions.

When adding THREAD_FLAG_MUTEX_UNLOCKED, I thought it might be useful to wait for multiple mutexes at the same time. IMO supporting this would mess up the mutex code, so I propose to remove the define.

gebart, 2017-10-11:
#7558 was merged, waiting for #7717 then this will be fixed.

PR #7717 (2017-10-13): core/thread_flags: Remove unused THREAD_FLAG_MUTEX_UNLOCKED
and improve docs for THREAD_FLAG_TIMEOUT

Fixes #7543
closed 2017-10-13 by gebart