RIOT-OS/RIOT #7560: iotlab-m3: samr21-xpro: shell not working

cgundogan, 2017-09-04:
When I flash an `iotlab-m3` or `samr21-xpro` board with `gnrc_networking` on current master, then I get no shell output. Flashing succeeds. `git log` indicates that there was some movement for cortex related code, but I haven't got the time to look further into in currently.

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 7.1.0
```

smlng, 2017-09-04:
cannot confirm that issue: tested with board samr21-xpro, pba-d-01-kw2x (aka PhyNode), and remote-revb on macOS with compiler:

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
```

PeterKietzmann, 2017-09-04:
Can confirm iotlab-m3 nodes are silent after flashing. samr21-xpro works as usual -for me. I'm using a rather old compiler `arm-none-eabi-gcc  4.9.3`

cgundogan, 2017-09-04:
@smlng you were right, the samr21-xpro works as expected. Wanna visit us downside and try with an iotlab-m3 node?

cgundogan, 2017-09-04:
BTW with `git checkout master@{one week ago}` the shell works like charm.

aabadie, 2017-09-04:
> BTW with git checkout master@{one week ago} the shell works like charm

Bisecting can help

smlng, 2017-09-04:
can confirm: samr21-xpro working, iotlab-m3 is not -> again macOS with gcc-6.3, as commented above

smlng, 2017-09-04:
bisected and got:

```
60be58ea6dc49dec386ab508df8f5994b6ce48a8 is the first bad commit
commit 60be58ea6dc49dec386ab508df8f5994b6ce48a8
Author: Vincent Dupont <vincent@otakeys.com>
Date:   Wed Aug 23 11:54:30 2017 +0200

    cpu/stm32_common: unify stmclk for stm32f[0|1|3] with f[2|4|7]

:040000 040000 c36d125b5c055c1e022cbad137964a67223dae84 1d4701f5e7427e4b856ff6d6d183c751eee053ea M	cpu
```

soleilos, 2017-09-04:
Tried gnc_networking in Board Nucleo F429 and the `ifconfig` also shows nothing on the shell. Is this related to this shell issue or?

aabadie, 2017-09-04:
#7561 proposes a fix. Thanks @smlng for pointing to the commit that introduced this issue.

smlng, 2017-09-04:
I don't have problem with `nucleo-f411` so it might be related to f1 only? @Tenderson but you get a shell, just the `ifconfig` command isn't showing anything? I suppose that's normal if no netdev is available.

Tenderson, 2017-09-04:
Thanks @smlng, gnrc_networking example works native very well. I need the IP address on the Nucleo board in this example. 

I followed [task 7.3](https://github.com/RIOT-OS/Tutorials/tree/master/task-07) with no success so far. After starting the udp server, I get no info from the ifconfig! Any thoughts is highly appreciated.

How can I enable netdev?

cgundogan, 2017-09-04:
@Tenderson this issues closed automatically because it was referenced in #7561 and it got merged. Is your issue still persisting?

closed 2017-09-04 by cgundogan

Tenderson, 2017-10-11:
@cgundogan I still have the issue :( . As I noted [#6629](https://github.com/RIOT-OS/RIOT/issues/6629) at the bottom. 

I appreciate any input from you too...

cgundogan, 2017-10-11:
@Tenderson what kind of transceiver did you attach and what driver do you use for the nucleo board?

Tenderson, 2017-10-12:
@cgundogan Can you be more specific? I simply build and flash my application on the fresh Nucleo144-f767 board and see no results from the shell.

I do the same on the Nucleo144-f429 board and it works well.

kYc0o, 2017-10-12:
@Tenderson have you tried to update the ST-Link firmware on your nucleo-f767?

Tenderson, 2017-10-13:
@kYc0o Yes, I updated the firmware with this tool, and still have the issue. Have you yourself tried on the nucleo144-F767?

haukepetersen, 2017-10-13:
@aabadie would you mind to give this board a quick test run with current master to verify the STDIO working? Thx.

aabadie, 2017-10-13:
@haukepetersen, I tested on iot-lab with current master and everything works fine.

@Tenderson, can you open a dedicated issue with a detailed description of your problem ? It will make things easier for people that want to help: AFAIU, your problem is related to nucleo-f767 which is not using the same CPU.

haukepetersen, 2017-10-13:
> I tested on iot-lab with current master and everything works fine.

Sorry, was not precise enough: could you test it on the `nucleo144-f767`? AFAIK you are the only one with access to that board in this round... 

But true, lets move this discussion in a dedicated issue for the `nucleo144-f767`...

aabadie, 2017-10-13:
Unfortunately, the nucleos are not me today. Maybe next week.

@Tenderson, with F7 there are problems with old versions of GCC. Which one are you using ?
I have this one:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

kYc0o, 2017-10-13:
> @haukepetersen, I tested on iot-lab with current master and everything works fine.

Yes, it was working fine since the PR which fixed the problem was merged.

>your problem is related to nucleo-f767 which is not using the same CPU.

Yes, for some reason we're using this issue to help (I think because we thought it was an easy issue), but true, it should have it's own.

Tenderson, 2017-10-16:
Hi @aabadie, 

I also thought this should be a quick fix. I will create another thread in this regard. I will do more tests and configs to see if I can resolve it myself, and will keep you guys posted. 

Thank you.

PR #7561 (2017-09-04): boards/iotlab-common: fix clock configuration
Should fix #7560.

The new clock configuration strategy has not been applied to the iotlab-xx boards. It works for me now.