RIOT-OS/RIOT #7562: GNRC_Networking example: ifconfig does not work!

Tenderson, 2017-09-04:
The gnrc_networking example works native very well. I need to setup a network of Nucleo boards.

I followed [task 7.3](https://github.com/RIOT-OS/Tutorials/tree/master/task-07) with no success so far on a specific board. After starting the udp server, I get no info from the `ifconfig`! Any thoughts is highly appreciated.

> The example just shows how to setup in native. The readme says: 
> This example shows you how to try out the code in two different ways: Either by communicating
> between the RIOT machine and its Linux host, or by communicating between two RIOT instances.
> Note that the former only works with native, i.e. if you run RIOT on your Linux machine.

Is there any guide on how to setup a network of real boards?

aabadie, 2017-09-04:
What kind if Nucleo boards are you using ? (Nucleo F429 with ethernet port ?)

smlng, 2017-09-04:
>  After starting the udp server, I get no info from the ifconfig! 

did you before starting the server?

smlng, 2017-09-04:
>  aabadie assigned smlng just now

nice one @aabadie 😉

Tenderson, 2017-09-04:
Thank you, Im using the Nucleo144- f429. Here is the log info:

Done flashing
/home/vagrant/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"
No handlers could be found for logger "root"
2017-09-04 13:42:59,079 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
help
2017-09-04 13:43:04,434 - INFO # help
2017-09-04 13:43:04,436 - INFO # Command              Description
2017-09-04 13:43:04,441 - INFO # ---------------------------------------
2017-09-04 13:43:04,446 - INFO # udp                  send data over UDP and listen on UDP ports
2017-09-04 13:43:04,447 - INFO # reboot               Reboot the node
2017-09-04 13:43:04,453 - INFO # ps                   Prints information about running threads.
2017-09-04 13:43:04,459 - INFO # ping6                Ping via ICMPv6
2017-09-04 13:43:04,460 - INFO # random_init          initializes the PRNG
2017-09-04 13:43:04,465 - INFO # random_get           returns 32 bit of pseudo randomness
2017-09-04 13:43:04,470 - INFO # ifconfig             Configure network interfaces
2017-09-04 13:43:04,476 - INFO # fibroute             Manipulate the FIB (info: 'fibroute [add|del]')
2017-09-04 13:43:04,483 - INFO # ncache               manage neighbor cache by hand
2017-09-04 13:43:04,484 - INFO # routers              IPv6 default router list
2017-09-04 13:43:04,489 - INFO # rpl                  rpl configuration tool ('rpl help' for more information)
ifconfig
2017-09-04 13:43:19,165 - INFO #  ifconfig
> udp server start 8888
2017-09-04 13:43:30,799 - INFO #  udp server start 8888
2017-09-04 13:43:30,805 - INFO # Success: started UDP server on port 8888
> ifconfig
2017-09-04 13:43:34,579 - INFO #  ifconfig
> routers
2017-09-04 13:43:41,887 - INFO #  routers
2017-09-04 13:43:41,888 - INFO # if  Router                          state      type
2017-09-04 13:43:41,888 - INFO # ---------------------------------------------------
>

aabadie, 2017-09-04:
I just flashed  `gnrc_networking` (latest master) on that board and have the expected behavior: `ifconfig` returns nothing, networking commands does nothing. This is normal since there's no support (yet) for the Ethernet port on this board.

There's a PR [here](#5643) that adds support for the STM32F4 Ethernet driver. With some effort it should be possible to have the networking commands work.

Are your interested in working on that ? Maybe @lebrush can give some help too.

Tenderson, 2017-09-04:
> I just flashed gnrc_networking (latest master) on that board and have the expected behavior: ifconfig returns nothing, networking commands does nothing. This is normal since there's no support (yet) for the Ethernet port on this board.

Could you tell me where it is pointed that the board does not support Ethernet port? 

> There's a PR here that adds support for the STM32F4 Ethernet driver. With some effort it should be possible to have the networking commands work.

The Link is not valid. I appreciate if you also let me know which ST Nucleo board supports the Ethernet module?

> Are your interested in working on that ? Maybe @lebrush can give some help too.

I am new to this and need to learn how this networking work.

aabadie, 2017-09-04:
> Could you please tell me where it is pointed that the board does not support Ethernet port?

For the nucleo boards with an Ethernet port (some of the nucleo 144 boards, f207, f429 and f746 are 2 of them), this is related to their MCU since some F4 support ethernet connectivity and the ethernet port is directly mapped to it. For F4, you can check if there is no `eth.c` file that in `cpu/stm32f4/periph` and this is what #5643 brings.

> The Link is not valid

Fixed (sorry)

Tenderson, 2017-09-04:
I could not find any `eth.c` in the Riot master. It means that actually Riot is not supporting any Ethernet module on its supported Boards.

lebrush, 2017-09-04:
The PR which provides ethernet support has not yet been merged. I'm working on an updated PR to the latest RIOT tip and I'll send tomorrow an update with an example so you can pull from there.

miri64, 2017-09-04:
> I could not find any eth.c in the Riot master. It means that actually Riot is not supporting any Ethernet module on its supported Boards.
>
> Is this right?

No, but at the moment there is only support for the [enc28j60](https://github.com/RIOT-OS/RIOT/tree/master/drivers/enc28j60) and [encx24j600](https://github.com/RIOT-OS/RIOT/tree/master/drivers/encx24j600) (and `netdev_tap` which is using the UNIX' `tap` interface to provide Ethernet support on native). If PRs for other drivers get merged, they will be supported as well.

Tenderson, 2017-09-05:
Thank you, @lebrush. I am really looking forward to your PR. I believe this is the most important thing for the Riot, so it can be deployed in real work, connecting devices and not only on native!

smlng, 2017-09-26:
Can we conclude that this is not an issue, at least not 

> GNRC_Networking example: ifconfig does not work!

but rather a missing feature, i.e. no driver for nucleo ethernet - but there is one in the pipe #5643 .

IMHO we can close this one, or definitely change the title!

miri64, 2017-09-26:
Yes.

closed 2017-09-26 by miri64