RIOT-OS/RIOT #7574: mrf24j40: NETOPT_STATE_STANDBY support

bergzand, 2017-09-06:
As far as I understand it, in the stand-by state, the radio should not receive packets, but be awake. Is it enough to simply turn off packet reception, or is a energy saving state expected? 
The mrf24j40 does not have a real stand-by mode where power consumption is minimized (a sleep mode is available). Is it enough to disable packet reception for NETOPT_STATE_STANDBY?

gebart, 2017-09-06:
While we always strive to use the lowest possible power mode, I think the main difference between sleep and standby is the time it takes the transceiver to switch to idle. I'd suggest that for devices such as the one you describe, it's better to leave the transceiver on but with paket reception disabled, than to make standby a synonym of sleep mode. One use case for standby is in MAC protocols where the radio needs to do a series of CCA checks or similar, before deciding whether to go to idle or go back to sleep. The timing may be important, hence it might not be the best choice to put the radio to sleep between CCA checks.

bergzand, 2017-09-06:
Thank you for your explanation. This is also what I was thinking, the real sleep mode of the mrf24j40 turns off the oscillator so that doesn't fit the requirement with the CCA check application.

I'll build a PR in the next days or so that disables packet reception on the mrf24j40 for this state.

closed 2017-09-06 by bergzand