RIOT-OS/RIOT #7585: Add support for ST b-l475e-iot01a board

aabadie, 2017-09-08:
Board datasheet is [here](http://www.st.com/en/evaluation-tools/b-l475e-iot01a.html), detailed pdf is [here](http://www.st.com/content/ccc/resource/technical/document/user_manual/group0/b1/b8/7a/f2/f7/8d/4b/6b/DM00347848/files/DM00347848.pdf/jcr:content/translations/en.DM00347848.pdf).

The board is based on STM32L475VG MCU.

Here is the list of missing features:

- [ ] I2C support for stm32l4
- [ ] ADC support for stm32l4
- [ ] driver for HTS221 (temperature and humidity) PR #7280 
- [ ] driver for LPS22HB (pressure)
- [ ] driver for VL53L0X (TOF and gesture detection)
- [ ] driver for MP34DT01 (microphone)
- [ ] driver for WiFi module Inventek ISM43362-M3G-L44
- [ ] driver for SPBTLE-RF (Bluetooth 4.1)
- [ ] driver for SPSGRF-868/915 (SubGH RF module)

aabadie, 2019-09-10:
Tracking issue, should not be closed.

miri64, 2019-09-10:
If they are marked as `Type: tracking`, stale bot will ignore them ;-). See https://github.com/RIOT-OS/RIOT/blob/master/.github/stale.yml#L7-L10