RIOT-OS/RIOT #7591: How to test enc28j60 driver?

holasm, 2017-09-10:
@haukepetersen how to test enc28j60 driver?

smlng, 2017-09-11:
For starters, you could use `tests/driver_enc28j60` with your favourite board, it has a SPI interface so wire it accordingly and adapt INT (interrupt) and RST (reset) pins in the test Makefile.

smlng, 2017-09-19:
any progress?

kbumsik, 2017-10-04:
Hi, I am working on #7433 and have a question too.
So while I run the test application and connect the two boards with a enc28j60 to a IPv6-enabled router then a board should be able to ping the other board in within the same subnet? And can the board also ping a Linux PC too?

maitysubhasis, 2017-10-17:
I have ensured that the SPI communication is working properly for my board (Openmote-cc2538).
Still the MAC address is showing 00:00:00:00:00:00. I am trying figure out what's the problem. Any have an idea what is going wrong?

haukepetersen, 2017-11-30:
This issue should be covered by #7745, so closing it.

closed 2017-11-30 by haukepetersen