RIOT-OS/RIOT #7605: samd21-xpro doesn't compile if gclk_src is not 0

photonthunder, 2017-09-14:
In the latest version or RIOT, uart.c was merged from SAML21 and SAMD21.  If you change the gclk_src from GCLK_CLKCTRL_GEN_GCLK0 to any other generic clock source it will not compile since the definitions in gclk.h are 32 bit rather than 8 bit.  @dylad any insight on the best way to fix this?

dylad, 2017-09-14:
@photonthunder The solution is to change `uint8_t gclk_src` -> `uint32_t gclk_src` in the file `cpu/sam0_common/include/periph_cpu_common.h`
I was able to reproduce this bug and this solution seems to fix it.
I'll open a PR to fix it right now. Thanks a lot for spotting this :)

photonthunder, 2017-09-14:
@dylad, happy to help.  Maybe it will help me get PR #7405 through, I tried to rebase it off your combination of uart.c.  Makes it nice to wake a board from sleep via uart.

dylad, 2017-09-14:
Nice effort, I'll take a look at it tomorrow ! I'll update my rework of I2C driver #7588 to prevent this issue too.
I just propose #7606 to fix the issue

photonthunder, 2017-09-14:
Perfect, closing...

PR #7606 (2017-09-15): cpu/sam0: fix sercom gclk selection
This PR should fix #7605 
If we try to use other values than GCLK_CLKCTRL_GEN_GCLK0 for UART (in boards/include/periph_conf.h), compilation fails (this issue only applies on SAMD21 based board but I cannot reproduce it for SAML21).
Thanks @photonthunder for spotting this :)
closed 2017-09-14 by photonthunder