RIOT-OS/RIOT #7614: Riot OS Ubuntu Dev Image

Tenderson, 2017-09-18:
Hi,

Is there any pre-configured Riot OS Ubuntu image for developers, so that we can start development immediately rather than going through error-prone steps [here](https://github.com/RIOT-OS/RIOT/wiki/Using-the-Eclipse-IDE-for-C-and-CPP-Developers,-Howto). 

It will be a great help to have such image with Eclipse, Flasher, Debugger configured. This saves a lot of time...

Thank you.

kbumsik, 2017-09-18:
There is a Vagrant config file in this repo that you can use. Simply `vagrant up` at the repository root directory.

Tenderson, 2017-09-19:
@kbumsik The vagrant file as you know does not setup the eclipse, it is just for the build environment. I have no problem with vagrant setup and can build and flash as well.

The problem is the Eclipse configuration for flashing and debugging, specially using the VBox. It would be great to have a configured Eclipse environment in Ubuntu, so the development can be started quickly..

smlng, 2017-09-20:
IMHO @kbumsik still has a point, if eclipse integration is a want-to-have feature we should adapt our vagrant box configuration instead of providing a separate VM box or something.

Tenderson, 2017-09-20:
Thank you @smlng, I do agree with you! 

This will be great help so the development can be started immediately.

basilfx, 2017-11-02:
For workshops I once created a Ubuntu Image Customizer. It basically takes the live image, removes crap, adds additional tools and generates a new ISO file. I've created a version that should install enough tools to run `make` for `examples/default`. You can find the customizer here: https://github.com/basilfx/RIOT-ubuntu-image.

To get started, download Ubuntu 16.04.3 desktop image, and follow the instructions in the `README.md` file.

This should not replace the Vagrant config file. It is just to share something I already made (took me 15 minutes to adapt). It can be useful for workshops where you don't want to bother the participants with setting up the environment and provide them a bootable live image.

Tenderson, 2017-11-02:
@basilfx Thank you, This is really helpful. I will check it in details. 

My main issue was to setup the debugger like GDB on the Eclipse for the Riot to debug programs on the ST Microcontrollers....

cgundogan, 2017-11-02:
@Tenderson Just FYI: I remember that we tried ~2 years ago a setup where we started a gdb server in the vagrant box and forwarded the port out of the box, so that an Eclipse running in the host system could connect to it.

cgundogan, 2017-11-02:
>IMHO @kbumsik still has a point, if eclipse integration is a want-to-have feature we should adapt our vagrant box configuration instead of providing a separate VM box or something.

hm, if general consensus is that we put more tools à la Eclipse into the box, then we definitely need an Ubuntu version with X/Wayland. Currently, we use the server edition without any graphical user interface.

PeterKietzmann, 2019-07-23:
Question about "Riot OS Ubuntu Dev Image" was answered.

closed 2019-07-23 by PeterKietzmann