RIOT-OS/RIOT #7616: Riot OS Ubuntu on VBox

Tenderson, 2017-09-18:
Hi, 

I am following the instructions [here](https://github.com/RIOT-OS/RIOT/wiki/Using-the-Eclipse-IDE-for-C-and-CPP-Developers,-Howto).

I am stuck at 7.2. for the both run settings:
 
1. At flashing, when I run it, I get an error: sudo: flash: command not found. However I can flash with the shell using OpenOCD as well. But in Eclipse, I face the above issue.

2. At Run configurations, I also get an error for recognizing the USB in VirtualBox: 
pyterm: error: unrecognized arguments: /dev/vboxusb/003 

In Vbox the usb format is not like: /dev/ttyUSB1 . 

How can I resolve these issues? Any thoughts is appreciated.

BytesGalore, 2017-09-19:
Hi @Tenderson, seems the instructions are getting a bit outdated.
But I'm happy you give it a try.
I cannot test it right now, but I guess you need to add `-p` or `--port` in front of the actually used port, e.g.:
``` 
pyterm -p /dev/vboxusb/003
```

smlng, 2017-09-19:
@Tenderson please report back and close issue if resolved, thx!

Tenderson, 2017-09-26:
I have other issues now. This should work now.

closed 2017-09-26 by Tenderson