RIOT-OS/RIOT #7641: Question: bitfield tests

travisgriggs, 2017-09-23:
In #7638, I really like the stylistic change to testing struct bitfields. E.g.

` while (!(dev(uart)->INTFLAG.reg & SERCOM_USART_INTFLAG_DRE)) {}`

vs

`while (!dev(uart)->INTFLAG.bit.DRE) {}`

This reads a ton better. I'm curious what the performance/size implications of this pattern are though? I did some naive tests and found that the code size didn't change at all. But that the performance was quite a bit faster for good old fashioned masking. I would assume that some of the bitfield manipulations would imply shifts in additions to masks and tests.

Has anyone (who is better at testing) verified that we're not paying too much of a price for this improved coding pattern?

smlng, 2017-09-23:
I'm not that much into compiler/code optimisation, but I would guess it should be faster, because compare ops can be optimised to use read-only regs.

gebart, 2017-09-23:
You need to check the assembly code for both solutions to know if there is a difference.

kbumsik, 2017-09-24:
I personally think it could be a little bit dangerous when portability matters more than the performance. The bit order of struct bitfield is not guaranteed by the compiler since it depends on the [CPU/Compiler endianess](http://mjfrazer.org/mjfrazer/bitfields/). Although it is true that the Big-Endian rarely can be found but it may lead confusion. I heard that the struct bitfield is intended for memery micro-optimization, not for bitwise operation.

PeterKietzmann, 2019-02-21:
@travisgriggs did you investigate on that?

closed 2019-09-25 by stale[bot]