RIOT-OS/RIOT #7642: Some broken builds on ubuntu 17.04

cladmi, 2017-09-24:
I wanted to run `dist/tools/compile_test/compile_test.py` on an ubuntu 17.04 machine and some configurations are broken:

* `tests/libfixmath_unittests` does not compile for `BOARD=native`
* `tests/ssp` does not compile for `BOARD=qemu-i386`

For `libfixmath_unittests/BOARD=native` I get the following error:
```
/home/harter/RIOT/tests/libfixmath_unittests/bin/pkg/native/libfixmath/libfixmath/fix16.h:274:9: error: left shift of negative value [-Werror=shift-negative-value]
         << 16U \
         ^
/home/harter/RIOT/tests/libfixmath_unittests/bin/pkg/native/libfixmath/unittests/libfixmath-unittests.h:7:11: note: in definition of macro ‘TEST’
     if (!(x)) { \
           ^
/home/harter/RIOT/tests/libfixmath_unittests/bin/pkg/native/libfixmath/libfixmath/fix16.h:310:11: note: in expansion of macro ‘FIXMATH_COMBINE_I_M’
         : FIXMATH_COMBINE_I_M(i, m) \
           ^~~~~~~~~~~~~~~~~~~
/home/harter/RIOT/tests/libfixmath_unittests/bin/pkg/native/libfixmath/unittests/fix16_macros_unittests.c:10:28: note: in expansion of macro ‘F16C’
   TEST(F16(i ## . ## m) == F16C(i,m)) \
                            ^~~~
/home/harter/RIOT/tests/libfixmath_unittests/bin/pkg/native/libfixmath/unittests/fix16_macros_unittests.c:102:3: note: in expansion of macro ‘DO_TEST’
   DO_TEST(  -1,07737)
   ^~~~~~~
```

And for `ssp/BOARD=qemu-i386` I get

```
/home/harter/RIOT/tests/ssp/bin/qemu-i386/tests_ssp.a(main.o): In function `test_func':
main.c:(.text.test_func+0x48): undefined reference to `__stack_chk_fail_local'
/usr/bin/ld: /home/harter/RIOT/tests/ssp/bin/qemu-i386/tests_ssp.elf: hidden symbol `__stack_chk_fail_local' isn't defined
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
/home/harter/RIOT/tests/ssp/../../Makefile.include:306: recipe for target 'link' failed
```

smlng, 2017-09-26:
@cladmi please provide gcc version you are using, further I think the qemu port is problematic anyway - but haven't tried that myself, maybe @kaspar030 knows something?!

cladmi, 2017-09-28:
Ubuntu 17.04 provides gcc-multilib version 6.3.0 which pulls

```
Package: gcc-multilib
Version: 4:6.3.0-2ubuntu1
Package: gcc
Version: 4:6.3.0-2ubuntu1
Package: gcc-6-multilib
Version: 6.3.0-12ubuntu2
Package: gcc-6
Version: 6.3.0-12ubuntu2
```

cladmi, 2017-09-28:
I upstreamed the issue for libfixmath https://github.com/PetteriAimonen/libfixmath/issues/3

cladmi, 2017-09-29:
For the `_stack_chk_fail_local` issue, duplicating `_stack_chk_fail` with the new name in ssp module removes the warning the test works.

https://github.com/RIOT-OS/RIOT/blob/67201c18f4e312bfa4bc2e198acf866ae3371af8/sys/ssp/ssp.c#L30

I will try to find what triggers the use of `_local` instead of the other one.

cladmi, 2017-09-29:
GCC defines `__stack_chk_fail_local` as a function calling `__stack_chk_fail` with some magic to save some code space

https://github.com/gcc-mirror/gcc/blob/6d191b11018a4346cdf17764a10d35e0515240b1/libssp/ssp-local.c

And it looks like that the `_local` function is used for static linking by using `ssp_nonshared`
http://wiki.osdev.org/Stack_Smashing_Protector#Usage
https://github.com/gcc-mirror/gcc/blob/6d191b11018a4346cdf17764a10d35e0515240b1/libssp/Makefile.am#L55

So maybe its currently a bug that `__stack_chk_fail` is used instead of `__stack_chk_fail_local`.

cladmi, 2018-01-05:
Board `qemu-i386` has been removed, so no more error for these two.

closed 2018-01-05 by cladmi