RIOT-OS/RIOT #7645: Build arduino bootloader compatible application

dylad, 2017-09-25:
I would like to know if there is some specific stuff to do in order to program an arduino m0 board (I know this board is not yet supported by RIOT). In fact, the Arduino M0 board does not have an embedded debugger. So I used BOSSA tool to flash my app on it. But each time, I ended by bricking the bootloader.
Arduino bootloader start at ROM address 0x00 with a 0x2000 bytes size.
Is there specific stuff to do in order to have a working app and keep a working bootloader ? (I don't want to use a hardware programmer each time i'm playing with this board...)
I looked for some documentation within RIOT but I didn't find anything on this subject.

gebart, 2017-09-25:
Did you try using the arduino mkr linker script (https://github.com/RIOT-OS/RIOT/blob/master/cpu/sam0_common/ldscripts/samd21g18a_mkr.ld)? It looks like there was already someone trying to get it working with the Arduino bootloader. The rom section starts at 0x2000, which should be safe for the bootloader. I don't have any time to dig further, but the linker script should be the first thing to check here.

Also examine the .map file from your build, it can be found in the bin/ dir next to the .elf file for your project. The .map should tell you at what address everything has been placed, try to find out where the interrupt vector has been placed. It should probably be located at 0x2000 for it to work.

aabadie, 2017-09-26:
> Did you try using the arduino mkr linker script

I confirm this is the way to go, I already used this one in #7510 and in other port not PR'ed yet. I opened #7646 in order to use a more generic name.

dylad, 2017-09-26:
Good to know, I already noticed this ld script but I was wondering if a SUBMODULE was needed or something to do with the Makefile in addition to this linked script.
@aabadie Did you successfully flash an app on a arduino board with a preinstalled bootloader using the current BOSSAC command build by RIOT ? (with make BOARD=<arduino_board> flash)
I also had some troubles on this point.

aabadie, 2017-09-26:
> Did you successfully flash an app on an arduino board with a preinstalled bootloader using the current BOSSAC command build by RIOT ?

Yes. The board has be set in "bootloader mode": on the boards I tested, this can be done by double pressing the reset button: the on-board led will then slowly oscillate.

You said that you tried to flash an Arduino M0 ? Is it the one with the original Arduino pinout layout ? If, yes, I don't think you need a bootloader and you can directly use OpenOCD (that's what I do). I have Arduino Zero and M0 pro.

dylad, 2017-09-26:
> on the boards I tested, this can be done by double pressing the reset button: the on-board led will then slowly oscillate

Ooooooh that did the trick ! I was able to flash successfully my arduino-m0 using BOSSAC.

I cannot (in fact I don't want to) use openOCD because I don't have an embedded debugger on the board like arduino Zero or M0-pro and I don't want to spend my time looking for my external programmer for openOCD.
the combination bootloader with bossac is really great.
BTW, I'm playing with this [board](http://robotdyn.com/catalog/boards/samd21_m0_mini/). which is very similar to the original. I can add support for `arduino-m0` to RIOT if you want.
My issue is solved. Thanks @aabadie & @gebart

ynezz, 2017-09-28:
> on the boards I tested, this can be done by double pressing the reset button: the on-board led will then slowly oscillate

FYI, Arduino for SAMD has quite [handy feature](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/USB/CDC.cpp#L118) in the Serial class, which allows putting board into bootloader mode automatically by just opening the serial port with 1200 baudrate setting. Reset feature as used in [Arduino IDE](https://github.com/arduino/Arduino/blob/ec2e9a642a085b32701cf81297ee7c1570177195/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L144).

So the reboot into bootloader and flashing can be done in two steps automatically:

```shell
$ stty -F /dev/ttyACM0 1200 && sleep 2
$ bossac ...
```

aabadie, 2017-09-28:
Well, that's what I tried initially (adapted from arduino-due) with mkr-1000, but couldn't make it work. I'll try again with your proposed patch in the other feather-m0 PR

dylad, 2017-09-28:
Thanks for the tips ! I'll give it a try soon or later...

ynezz, 2017-09-28:
I've just realized, that to get this `reboot into bootloader ` feature working in RIOT, there would need to be USB/CDC driver available first. In other words, it is going to work for the first time, if your board is running some sketch with Arduino Serial class using USB/CDC serial driver which has this `reboot into bootloader` feature implemented. Once you flash your device with RIOT based firmware, you can't use this feature anymore.

Anyway, it might still be possible to emulate [this feature](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/Reset.cpp) in RIOT using `bootloader` shell command for example.

aabadie, 2017-09-28:
> Once you flash your device with RIOT based firmware, you can't use this feature anymore.

Exact, I tried again and came up to the same conclusion. Having USB/CDC support to RIOT will also allow to redirect STDIO to USB, which is very useful for debugging without a FDTI.

gebart, 2017-09-28:
For reference, there was an in-progress USB stack in #3890, but it was never finished. In my opinion, a USB CDC driver would be an incredibly useful feature addition.

aabadie, 2017-09-30:
I'm going to close this one, since the initial issue is solved. Feel free to reopen if you don't agree.

closed 2017-09-30 by aabadie