RIOT-OS/RIOT #7649: RFC: (TLSMAN) API for handling all the (D)TLS stacks for the user

rfuentess, 2017-09-27:
This is the proposition of the second API (See #7397) for handling secure communication with RIOT O.S. This API has as objective to binding all the SSL/TLS protocols available (tinyDTLS, WolfSSL, etc.) with the intention of making their use invisible to the user.  With this approach the user only needs to use secure_sock and configure the (D)TLS stack by means of the Makefile.

I had the good luck of meeting @kaleb-himes during the RIOT summit and we had a brief discussion about the feasibility of this approach for the [tinyDTLS](https://github.com/RIOT-OS/RIOT/pull/7615 ) and [WolfSSL](https://github.com/RIOT-OS/RIOT/pull/7348) stacks.

General scheme
----
```
---------------------------------------------------
|                                                 |
|    Application / Library                        |
|                                                 |
---------------------------------------------------
                |
                |   secure_sock_api
---------------------------------------------------
|                                                 |
|    Secure_Sock                                  |
|                                                 |
|          ----------------------------------     |
|          |  TLSMAN                        |     |
|          |                                |     |
|          |  -------------   ------------- |     |
|          |  | tinydtls  |   | WolfSSL   | |     |
|          |  |           |   |           | |     |
|          |  -------------   ------------- |     |
|          ----------------------------------     |
---------------------------------------------------
        |                   |
        | sock_tcp          | sock_udp
        |                   |
        .                   .
        .                   .
        .                   .
```

API: TLSMAN ((D)TLS stack Manager)
-----

API used directly by secure_sock for handling the communication between the
TLS/DTLS stack and the corresponding sock_tcp and sock_udp. 

```C

 /* Structure defined according to the (D)TLS */
typedef struct {
     ...
     remote_peer
     local_peer
     secure_sock_key_peer_t keys
     union sock {
                 sock_upd udp; 
                 sock_tcp tcp
               }
     wolf_session
     tinydtls_session
} tlsman_session_t

/*
 * Load the TLS/DTLS stack(s) available.
 * TODO: Handling the selection of cipher suites at this point.
 */
size_t tlsman_load_stack(int **ciphers)

/*
 * Start a DTLS handshake between the peers defined by session.
 *
 * NOTE: This can be optional 
 *
 * @session The secure session to use.
 * @flags   If SECURE_SOCK_NONBLOCK is set, this function ends after sending
 *          the first handshake flight. Otherwise, blocks until the handshake
  *         is completed.
 * @return: 0 on success, otherwise error code.
*/
size_t tlsman_client_open_channel(tlsman_session_t *session, unsigned int flags)

/*
 * Starts listening for any (D)TLS record message.
 * NOTE: See tlsman_retrieve_data_app() for the Data App record.
 * @session The secure session to use.
 * @flags   If SECURE_SOCK_NONBLOCK is set (default) the function ends after
 *          sending the first handshake flight. Otherwise, blocks until the
 *          handshake is completed.
 * @return: 0 on success, otherwise error code.
*/
tlsman_server_open_channel(tlsman_session_t *session, unsigned int flags,
                           secure_sock_key_peer_t keys )

/*
 * Returns the last error occurred in the local peer (Alerts, retransmission,
*  wrong cipher suites,  etc.).
 * @return: Error code (0 if nothing)
 */
ssize_t tlsman_return_last_known_error()

/*
 * Process the @error_id and identifies if the (D)TLS session must be terminated.
 * @error_id  Error ID
 * @return 0 if error is non-fatal.
 */
ssize_t tlsman_process_error_code (site_t error_id)

/*
 * Checks if there is a (D)TLS Data Application record pending to be passed to
 * secure_sock.  
 *
 * NOTE: This can be optional 
 *
 * @returns 0 if not.
 */
size_t tlsman_is_dtls_data_app_ready( )

/*
 * Process the last (D)TLS Data Application record available.
 * @session  The secure session to use.
 * @data     Pointer where the received data should be stored.
 * @max_len  Maximum space available at  @data.
 * @secure_handler  Callback for handling the upper layer data (can be NULL)
 * @returns  Number of bytes retrieved or error code.
 */
ssize_t tlsman_retrieve_data_app(tlsman_session_t *session,
                                 void *data,
                                 size_t max_len,
                                 sec_data_handler secure_handler )

/*
 * Send upper layer data as a (D)TLS Data Application record.
 * @session  The secure session to use.
 * @data     Pointer to data to send
 * @data_size Size of  @data.
 * @returns  0 on success or error code.
 */
size_t tlsman_send_data_app(tlsman_session_t *session, const void *data,
                            size_t data_size )

/*
 * Close the (D) TLS channel established between the peers of the session.
 * This also sends a last DTLS Alert record and releases memory used for the
 * session.
 * @returns  0 on success or error code.
*/
size_t tlsman_close_channel(tlsman_session_t *session)

```

The API can be configured to work with synchronous or asynchronous being the latest the default setting. In the case of `tlsman_retrieve_data_app ()` it should be called by `secure_sock_send ()` when asynchronous is used and by `secure_sock_rcvd ()` otherwise. 

For the first PR the support will be tested only for tinyDTLS because WolfSSL is still being integrated to RIOT.

---

Any suggestions or comments? (like better names for this API)

rfuentess, 2018-04-09:
I had been rethinking the two proposed modules for trying to make them the most independent possible from network stacks.  I'm moving the most relevant aspects of the sock secure module to TLSMAN (flags, cipher suites selection, open channel, renegotiation, etc.). With the idea that sock secure be a module only for socks, and that other modules can be developed for different stacks. 

Right now I have a client side test working with tinyDTLS (only in native), which can also support the asynchronous sock (PR #8236). I want to generate a second client test using sockets (as tinyDTLS can support them) to show the feasibility of changing between those two stacks, and of course testing on real hardware.

For now, I'm  updating the API for TLSMAN.

```c
typedef struct _sock_tl_ep tlsman_ep_t; /**< Endpoint for a tlsman object */

/* Control flags for the (D)TLS stacks */
#define TLSMAN_FLAG_STACK_UNIVERSAL (0x00) /**< Loads the default (D)TLS stack */
#define TLSMAN_FLAG_STACK_TLS (0x01) /**< Only loads TLS stack */
#define TLSMAN_FLAG_STACK_DTLS (0x02) /**< Only loads DTLS stack */
#define TLSMAN_FLAG_STACK_IPSEC (0x03) /**< Only loads IPSEC @NOTE not supported yet */
#define TLSMAN_FLAG_NONBLOCKING (0x10) /**< Define [non-]blocking behavior @NOTE If supported */

/* Maximum number of attempts for completing a (D)TLS Handhsake */
#define TLSMAN_HANDSHAKE_MAX_TRY 10

/**
 * Error codes
 */
typedef enum{
    TLSMAN_ERROR_SUCCESS = 0x00,            /**< Non-error */
    TLSMAN_ERROR_FATAL = 0x01,              /**< Fatal error (non specified below) */
    TLSMAN_ERROR_UNABLE_CONTEXT,            /**< Unable to create context */
    TLSMAN_ERROR_TRANSPORT_NOT_SUPPORTED,   /**< TCP/UDP stack selected not supported  */
    TLSMAN_ERROR_CIPHER_NOT_SUPPORTED,      /**< Cipher suite(s) selected not supported */
    TLSMAN_ERROR_HANDSHAKE_TIMEOUT,         /**< Timeout for the handshake (first negotiation) */
    TLSMAN_ERROR_DATA_APP_WRITE,            /**< Unable to send a DTLS Data App record   */
    TLSMAN_ERROR_SESSION_REN,               /**< /*TODO: Separated both types of re-negotiation? */
    TLSMAN_ERROR_NON_STARTED                /**< Non secure session has ben done yet */

} tlsman_errors_t;

/**
 * @brief  Abstract (D)TLS secure session between peers.
 *
 */
typedef struct {

#ifdef MODULE_TLSMAN_TINYDTLS
    /**
     * @brief The tinyDTLS context for handling the secure session.
     *
     * @note only avaliable if the submodule secure_sock_tinydtls is defined.
     */
     tinydtls_session_t tinydtls_context;
#endif
#ifdef MODULE_TLSMAN_WOLFSSL
    /**
     * @brief The WolfSSL context for handling the secure session.
     *
     * @note only avaliable if the submodule secure_sock_wolfssl is defined.
     */
    wolffssl_session_t wolfssl_context;
#endif

} tlsman_session_t;

/**
 *  @brief Initializes the (D)TLS stack loaded.
 *
 * Loads the initialization of the (D)TLS stack. It's expected to be only one
 * compiled stack (tinydtls, wolfssl, etc.).
 *
 * @note Not all the (D)TLS stacks can support loading cipher suites (e.g.
 * tinyDTLS). In such cases, is only checked that the stack was compiled with
 * one or more of the cipher suites provided in cipersuites_list.
 * @param[in]  ciphersuites_list An array of cipher suites ID to load
[... 151 lines cut ...]
 * @param[in]  flags    Flags for defining the [nonb-]blocking behavior.
 * @param[in]  pkt_buff If nonb-bloking, pointer to packet buffer to be used. Otherwise can be NULL
 * @param[in]  pkt_size Size of pkt_buff
 *
 * @return 0 on success.
 * @return TLSMAN_ERROR_SESSION_REN if renegotiation failed.
 */
ssize_t tlsman_renegotiate_session(tlsman_session_t *session, uint8_t flags,
                                    uint8_t * pkt_buff, size_t pkt_size);

/**
 *  @brief Generates a new (D)TLS handshake for the current secure session.
 *
 * @param[in]  session  The secure session to use.
 * @param[in]  flags    Flags for defining the [nonb-]blocking behavior.
 * @param[in]  pkt_buff If nonb-bloking, pointer to packet buffer to be used. Otherwise can be NULL
 * @param[in]  pkt_size Size of pkt_buff
 *
 * @return 0 on success.
 * @return TLSMAN_ERROR_SESSION_REN if renegotiation failed.
 */
ssize_t tlsman_new_handshake(tlsman_session_t *session, uint8_t flags,
                                   uint8_t * pkt_buff, size_t pkt_size);

```

miri64, 2019-02-01:
See https://github.com/RIOT-OS/RIOT/pull/9450#issuecomment-459733076

closed 2019-02-01 by miri64