RIOT-OS/RIOT #7661: stm32_common pm.c

paula75, 2017-09-29:
Working on pm feature for stm32l1 i noticed pm.c is on stm32_common instead of each of the different folders for every processor. The problem with this is that each processor have different amount of pm modes and need to set different register to get to them, this result in use a lot of #if defined in the functions. Is there a special reason to implement this module in stm32_common instead of each processor separately like the other periph such as adc, i2c, etc?

roberthartung, 2017-10-20:
stm32l1 uses both: stm32_common and cortexm_common. As far as I understand, the MCUs might be similar and should therefore be implemented in stm32_common or cortexm_common. However, if you have a use case where too much #ifdef is used and you can show that it would be a good choice to separate pm.c for the different cpus (stm32l1, stm32f4, ..), go ahead and propose it in a new PR. But until then I would like to keep the pm.c in stm32_common and/or cortexm_common to avoid too many files with almost the same code.
The drawback might be that we have to decide on a common number of modes that can be used.. Do you have anything spefific in mind? is there a WIP PR?

temmihoo, 2017-11-14:
#7734 is related

closed 2019-09-10 by stale[bot]