RIOT-OS/RIOT #7672: 802.15.4 support for nRF52840 MCU

hubertmis, 2017-10-04:
I've seen, there is support for nRF52840 PDK board in RIOT-OS.

Recently we released driver for IEEE 802.15.4 radio in nRF52840: https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver
I think it would be good to add 802.15.4 support to nRF52840 RIOT port.

haukepetersen, 2017-10-05:
Thanks for your input, so nice to see the device driver to be published as open source!

Support for the 802.15.4 mode of the nRF52840 radio is indeed something we definitely want in the RIOT master, and we actually started to write a driver from scratch some months ago. The driver is almost ready and close to be merged -> #7425

Naturally the RIOT implementation is quite entangled with RIOT specific interfaces (e.g. `netdev`). I think it would be very nice to look a little closer into both implementations and see how they compare or if it makes sense for us to go with Nordic's implementation. @dnahm Would you be willing to take a look?

dnahm, 2017-10-05:
I think we should integrate this radio driver in RIOT, because it provides all features.

hubertmis, 2017-10-05:
It's good to know that there is almost ready driver for nRF52840 radio. I didn't know there is #7425.
The driver we released is heavily tested and robust. It will be easier to test one implementation instead of two. Moreover, it is fully event-driven (no busy waits for events). It should result in lower power consumption.

Zhiyi-Zhang, 2018-11-07:
Any new updates on this issue? We are really looking forward to this feature to be merged.

bergzand, 2019-03-25:
This can be closed now that #10268 is merged.

closed 2019-03-25 by bergzand