RIOT-OS/RIOT #7698: Redefinition of iovec with current glibc

pyropeter, 2017-10-09:
```
$ git describe 
2017.10-devel-604-g6da78131e
$ cd examples/default/
$ make clean all
Building application "default" for "native" with MCU "native".

"make" -C $RIOT/boards/native
"make" -C $RIOT/core
"make" -C $RIOT/cpu/native
"make" -C $RIOT/drivers
"make" -C $RIOT/sys
"make" -C $RIOT/boards/native/drivers
"make" -C $RIOT/cpu/native/netdev_tap
"make" -C $RIOT/drivers/netdev_eth
"make" -C $RIOT/sys/auto_init
"make" -C $RIOT/sys/auto_init/netif
"make" -C $RIOT/drivers/saul
"make" -C $RIOT/cpu/native/periph
"make" -C $RIOT/cpu/native/vfs
In file included from /usr/include/sys/socket.h:26:0,
                 from /usr/include/net/if.h:26,
                 from $RIOT/cpu/native/include/netdev_tap.h:35,
                 from $RIOT/cpu/native/include/netdev_tap_params.h:22,
                 from $RIOT/sys/auto_init/netif/auto_init_netdev_tap.c:24:
/usr/include/bits/types/struct_iovec.h:26:8: error: redefinition of ‘struct iovec’
 struct iovec
        ^~~~~
In file included from $RIOT/drivers/include/net/netdev.h:198:0,
                 from $RIOT/cpu/native/include/netdev_tap.h:28,
                 from $RIOT/cpu/native/include/netdev_tap_params.h:22,
                 from $RIOT/sys/auto_init/netif/auto_init_netdev_tap.c:24:
$RIOT/sys/libc/include/sys/uio.h:33:8: note: originally defined here
 struct iovec {
        ^~~~~
make[4]: *** [$RIOT/Makefile.base:81: $RIOT/examples/default/bin/native/auto_init_gnrc_netif/auto_init_netdev_tap.o] Error 1
make[3]: *** [$RIOT/Makefile.base:20: ALL--$RIOT/sys/auto_init/netif] Error 2
make[2]: *** [$RIOT/Makefile.base:20: ALL--$RIOT/sys/auto_init] Error 2
make[1]: *** [$RIOT/Makefile.base:20: ALL--$RIOT/sys] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [$RIOT/examples/default/../../Makefile.include:306: link] Error 2
```

nmeum, 2017-10-18:
Any updates on this? Isn't this a critical issue for the upcoming release?

miri64, 2017-10-25:
> Any updates on this? Isn't this a critical issue for the upcoming release?

It seems like most people testing for the release use older versions of glibc. @nmeum @pyropeter @cgundogan can you test with #7849?

PR #7849 (2017-10-26): sys: libc: only add include path on non-native boards
Tries to fix #7698 where an update to glibc causes this include path to collide with definitions in glibc's include path.
closed 2017-10-26 by cgundogan