RIOT-OS/RIOT #7701: Z1 Zolertia tests/driver_adxl345 failed: unsatisfied feature requirements: periph_i2c

XiumingLiu, 2017-10-09:
Hello, 

I am new to RIOT and was going through some tests with a Z1 Zolertia node. The test for adxl345 sensor failed with the following output: 

wcnes@wcnes-VirtualBox:~/RIOT/tests/driver_adxl345$ make BOARD=z1
There are unsatisfied feature requirements: periph_i2c

EXPECT ERRORS!

Building application "tests_driver_adxl345" for "z1" with MCU "msp430fxyz".

"make" -C /home/wcnes/RIOT/boards/z1
"make" -C /home/wcnes/RIOT/core
"make" -C /home/wcnes/RIOT/cpu/msp430fxyz
"make" -C /home/wcnes/RIOT/cpu/msp430_common
"make" -C /home/wcnes/RIOT/cpu/msp430_common/periph
"make" -C /home/wcnes/RIOT/cpu/msp430fxyz/periph
"make" -C /home/wcnes/RIOT/drivers
"make" -C /home/wcnes/RIOT/drivers/adxl345
"make" -C /home/wcnes/RIOT/drivers/periph_common
"make" -C /home/wcnes/RIOT/sys
"make" -C /home/wcnes/RIOT/sys/auto_init
"make" -C /home/wcnes/RIOT/sys/div
"make" -C /home/wcnes/RIOT/sys/isrpipe
"make" -C /home/wcnes/RIOT/sys/oneway-malloc
"make" -C /home/wcnes/RIOT/sys/tsrb
"make" -C /home/wcnes/RIOT/sys/uart_stdio
"make" -C /home/wcnes/RIOT/sys/xtimer
/home/wcnes/RIOT/tests/driver_adxl345/bin/z1/adxl345.a(adxl345.o): In function `adxl345_init':
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:50: undefined reference to `i2c_acquire'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:53: undefined reference to `i2c_init_master'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:54: undefined reference to `i2c_release'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:60: undefined reference to `i2c_read_reg'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:62: undefined reference to `i2c_release'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:67: undefined reference to `i2c_write_regs'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:70: undefined reference to `i2c_write_reg'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:71: undefined reference to `i2c_write_reg'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:73: undefined reference to `i2c_write_reg'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:76: undefined reference to `i2c_release'
/home/wcnes/RIOT/tests/driver_adxl345/bin/z1/adxl345.a(adxl345.o): In function `adxl345_read':
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:89: undefined reference to `i2c_acquire'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:90: undefined reference to `i2c_read_regs'
/home/wcnes/RIOT/drivers/adxl345/adxl345.c:91: undefined reference to `i2c_release'
collect2: ld returned 1 exit status
/home/wcnes/RIOT/tests/driver_adxl345/../../Makefile.include:306: recipe for target 'link' failed
make: *** [link] Error 1

If we look into the  RIOT/boards/z1/Makefile.features file, we won't see the following line

FEATURES_PROVIDED += periph_i2c

And there is no I2C configuration in the  RIOT/boards/z1/include/periph_conf.h file.

Does it mean that currently i2c is not defined for Z1?

Many thanks!

dylad, 2017-10-10:
Hi @xiumingliu and welcome to RIOT !
Looking at cpu/msp430fxyz, it seems like i2c is not supported by RIOT yet. If you want to used your Z1 board you have to add support for i2c first.

kaspar030, 2017-10-23:
There's a software i2c implementation for MSP430 that you could give a try in #5415.

I'll close this for now, feel free to reopen if more questions arise.

closed 2017-10-23 by kaspar030