RIOT-OS/RIOT #7713: Findings from testing NDP2

smlng, 2017-10-10:
I started testing NDP2 based on #7479, this issue is meant to collect observations, problems and findings observed, so far. List will be extended

## General observations

- no DAD what so ever
- no MLD, for joins on `ff02::1` and `ff02::2`

## IPv6 standard ND

tested between RIOT native (with modified `examples/gnrc_networking`) and host system (macOS, or Linux)

**RIOT as host**
- sends RS every 4s, while the interval is according to RFC 4861, RFC default would be 3 retransmits only, not indefinitely.
- looses (or does not generate at all) a IPv6 autogenerated address from RA PIO send by `radvd` from Linux host, when PIO L (on-link) flag is set (on) - on current master with ndp1 it works
- if PIO L (on-link) flag is 0/off it generates a global IPv6 address from that prefix, address persists
- however prefix is not shown on `nib prefix` and also not in `nib route`, the latter has only the default entry, which is likely sufficient

**RIOT as router**

- RAs initially send in short interval, which is correct but only 2 RAs instead of 3 as proposed by RFC
- subsequent RAs are roughly send periodically every 500s which should be fine
- prefix lifetime in PIOs is `infinity` by default, might be better to have a lower default?

## IPv6 6LoWPAN ND

smlng, 2017-10-10:
failed assertion using `gnrc_networking` adapted Makefile to have `USEMODULE += gnrc_ipv6_nib_6ln`, setting global address `fd16::1/64` on node A and `fd16::2/64` on node B and try ping6 A -> B:

```
2017-10-10 22:55:21,645 - INFO #  ifconfig 7 add fd16::1/64
2017-10-10 22:55:21,648 - INFO # success: added fd16::1/64 to interface 7
> ifconfig
2017-10-10 22:55:24,240 - INFO #  ifconfig
2017-10-10 22:55:24,248 - INFO # Iface  7  HWaddr: 34:a6  Channel: 26  Page: 0  NID: 0x23
2017-10-10 22:55:24,251 - INFO #           Long HWaddr: 79:65:36:4c:70:34:34:a6 
2017-10-10 22:55:24,256 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2017-10-10 22:55:24,262 - INFO #           ACK_REQ  CSMA  MTU:1280  HL:64  Source address length: 8
2017-10-10 22:55:24,265 - INFO #           Link type: wireless
2017-10-10 22:55:24,271 - INFO #           inet6 addr: fe80::7b65:364c:7034:34a6  scope: local  VAL
2017-10-10 22:55:24,275 - INFO #           inet6 addr: fd16::1  scope: global  VAL
2017-10-10 22:55:24,276 - INFO #           
> nib prefix
2017-10-10 22:55:49,399 - INFO #  nib prefix
2017-10-10 22:55:49,401 - INFO # fd16::/64 dev #7 
> nib route
2017-10-10 22:55:52,879 - INFO #  nib route
2017-10-10 22:55:52,881 - INFO # fd16::/64 dev #7
> ping6 fd16::2
2017-10-10 22:56:04,723 - INFO #  ping6 fd16::2
2017-10-10 22:56:05,723 - INFO # ping timeout
2017-10-10 22:56:06,723 - INFO # 0x39bf
2017-10-10 22:56:06,725 - INFO # *** RIOT kernel panic:
2017-10-10 22:56:06,727 - INFO # FAILED ASSERTION.
2017-10-10 22:56:06,728 - INFO # 
2017-10-10 22:56:06,734 - INFO # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2017-10-10 22:56:06,742 - INFO # 	  - | isr_stack            | -        - |   - |    512 (  200) | 0x20000000 | 0x200001b8
2017-10-10 22:56:06,750 - INFO # 	  1 | idle                 | pending  Q |  15 |    256 (  128) | 0x2000054c | 0x200005cc 
2017-10-10 22:56:06,758 - INFO # 	  2 | main                 | pending  Q |   7 |   1536 (  904) | 0x2000064c | 0x20000a14 
2017-10-10 22:56:06,766 - INFO # 	  3 | pktdump              | bl rx    _ |   6 |   1536 (  248) | 0x20003074 | 0x2000357c 
2017-10-10 22:56:06,774 - INFO # 	  4 | 6lo                  | bl rx    _ |   3 |   1024 (  340) | 0x20003678 | 0x2000395c 
2017-10-10 22:56:06,783 - INFO # 	  5 | ipv6                 | running  Q |   4 |   1024 (  744) | 0x20001178 | 0x20001444 
2017-10-10 22:56:06,791 - INFO # 	  6 | udp                  | bl rx    _ |   5 |   1024 (  276) | 0x20003c00 | 0x20003eec 
2017-10-10 22:56:06,799 - INFO # 	  7 | at86rf2xx            | bl rx    _ |   2 |   1024 (  456) | 0x20000d38 | 0x2000102c 
2017-10-10 22:56:06,805 - INFO # 	    | SUM                  |            |     |   7936 ( 3296)
2017-10-10 22:56:06,805 - INFO # 
2017-10-10 22:56:06,806 - INFO # *** halted.
2017-10-10 22:56:06,806 - INFO # 
```

miri64, 2017-10-10:
> * no DAD what so ever
* no MLD, for joins on ff02::1 and ff02::2

It's a little bit unfair to point this out if that was never the goal for NIB (not NDP2, that's just the send and header assembly module now).

miri64, 2017-10-10:
> * prefix lifetime in PIOs is infinity by default, might be better to have a lower default?

Did you add the prefix using `ifconfig <if> add` or `nib prefix add`? With the latter you can configure the lifetime.

miri64, 2017-10-10:
(but until RFC 4862 isn't implemented properly this isn't important anyway).

miri64, 2017-10-10:
Rest I will fix tomorrow :-)

smlng, 2017-10-10:
> It's a little bit unfair to point this out if that was never the goal for NIB

wasn't meant to diss NDP2 by that, more as a general ToDo what RIOT is missing, sorry if you got that wrong.

miri64, 2017-10-10:
Did not got it wrong, just wanted to make sure we are on the same page here ;-)

miri64, 2017-10-11:
Just to clarify: the name of the new neighbor discovery implementation is not NDP2! `gnrc_ndp2` is just the packet assembly module for NDP packets (I sadly wasn't able to use the old `gnrc_ndp` part for that). The actual handling of NDP packets its part of the NIB.

miri64, 2017-10-11:
> * sends RS every 4s, while the interval is according to RFC 4861, RFC default would be 3 retransmits only, not indefinitely.

Just looked into this: The indefinite part comes from [RFC 6775](https://tools.ietf.org/html/rfc6775#section-5.3). I'll fix it for classic IPv6.

miri64, 2017-10-11:
>> * sends RS every 4s, while the interval is according to RFC 4861, RFC default would be 3 retransmits only, not indefinitely.

> Just looked into this: The indefinite part comes from [RFC 6775](https://tools.ietf.org/html/rfc6775#section-5.3). I'll fix it for classic IPv6.

Fixed in 9b77bb7

miri64, 2017-10-11:
> - looses (or does not generate at all) a IPv6 autogenerated address from RA PIO send by `radvd` from Linux host, when PIO L (on-link) flag is set (on) - on current master with ndp1 it works
> - if PIO [...]

I'm unsure how you mean that. Do you expect a set L-flag to auto-generate an address (that would be wrong, a set A flag does that) or are you pointing out that with an set A flag but an unset L flag, the address is auto-generated, while a set A flag and a set L flag doesn't?

miri64, 2017-10-11:
> * RAs initially send in short interval, which is correct but only 2 RAs instead of 3 as proposed by RFC

Also unclear, how you mean that: do you mean there are *only* 2 RAs send in short intervals and then none afterwards (which would be wrong) or are the only 2 RAs in very short intervals and then longer intervals between them afterwards (which except for the 2 be the right way).

miri64, 2017-10-11:
@smlng also, I get 4 (!) RAs in 16s interval differences (which is the specified short interval). Can you maybe provide a dump of the output with the following patch?

```patch
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/_nib-router.c b/sys/net/gnrc/network_layer/ipv6/nib/_nib-router.c
index a3e4675..dafdd58 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/_nib-router.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/_nib-router.c
@@ -69,6 +69,8 @@ void _handle_snd_mc_ra(gnrc_netif2_t *netif)
             }
             /* netif->ipv6.ra_sent overflowed => this was our last final RA */
             if (netif->ipv6.ra_sent != 0) {
+                printf("test %u (next in %u ms)\n", netif->ipv6.ra_sent,
+                       next_ra_time);
                 _evtimer_add(netif, GNRC_IPV6_NIB_SND_MC_RA, &netif->ipv6.snd_mc_ra,
                              next_ra_time);
             }
```

smlng, 2017-10-11:
> with an set A flag but an unset L flag, the address is auto-generated, while a set A flag and a set L flag doesn't?

exactly, the A flag is set in both cases but an address is only auto generated if L flag is **not** set, from what I read the L flag should not have any influence on whether or not an IPv6 address is auto generated.

smlng, 2017-10-11:
>  are the only 2 RAs in very short intervals and then longer intervals between them afterwards (which except for the 2 be the right way).

also right, I see 2 sort RAs (namely at 1.5 and 16.2 seconds, for instance) and the next after roughly 500s and 1000s and so on (every 500s, approx). The standard defines 3 RAs retransmission in short succession when a link gets ready, however that is (and should be) configurable. I think its set to 3 but the comparison is _wrong_ so only 2 are actually send in short interval on start up.

smlng, 2017-10-11:
> I get 4 (!) RAs in 16s interval differences 

hmm, maybe I tested with an older version, will retest - but the general behaviour is already right, just the actual count seemed of in my tests.

miri64, 2017-10-11:
Can you also reply to https://github.com/RIOT-OS/RIOT/issues/7713#issuecomment-335607482?

smlng, 2017-10-11:
> Just looked into this: The indefinite part comes from RFC 6775.

yes but it also states:

_The RECOMMENDED rate of retransmissions is to
   initially send up to 3 (MAX_RTR_SOLICITATIONS) RS messages separated
   by at least 10 seconds (RTR_SOLICITATION_INTERVAL) as specified in
   [RFC4861], and then switch to slower retransmissions._

what I saw was indefinite RS with an 4s interval in my tests, above it reads 10s for the first 3 and **slower** interval (that is >10s) afterwards.

miri64, 2017-10-13:
> what I saw was indefinite RS with an 4s interval in my tests, above it reads 10s for the first 3 and slower interval (that is >10s) afterwards.

Yes for classic NDP it is 4s. I only stated the behavior was taken from 6LoND, not the timings ;-)

miri64, 2017-10-13:
>> I get 4 (!) RAs in 16s interval differences
>
> hmm, maybe I tested with an older version, will retest - but the general behaviour is already right, just the actual count seemed of in my tests.

@smlng have you?

miri64, 2017-11-16:
@smlng how do we want to handle this issue? Do you want to add your further findings here? Or in #7925? Do you want to test #7479 and report your findings and I fix them there or should we #7479 first and we centralize all work in #7925?

miri64, 2017-11-24:
As I did not see the reported RAs and @bergzand also didn't report anything like this, I close this issue for now. Please reopen if you disagree.

smlng, 2017-11-27:
Started (re)testing of #7925 😄 

First issues I found were a compile error and a command which wasn't working as expected see #8144.

Further, I found the following behaviour I'm unsure if that is correct or expected, here is what I did:

I run single instance of `gnrc_networking` on native (macOS) and do some pings to the host system and use Wireshark on the tap bridge:

1. `ping6 ff02::1` from RIOT, I see ping messages and NS from host for RIOT nodes lladdr
2. `ping6 fe80::<host addr>` from RIOT to lladdr of host, I see NS from RIOT on sol. mcast address as expected and also the NA
3. I delete the now existing nib neigh entry for the host lladdr, and afterwards I manually add it again using `nib neigh add 7 fe80::<host addr> <lladdr>`
4. again ping6 as in 2, but I still see NS on sol.mcast addr for the host addr before the pings, though I would expect that NS/NA are omitted as I manually made a nib entry for that IP

smlng, 2017-11-27:
forgot to clarify: 1 and 2 look fine to me, but 4 seems to be odd

bergzand, 2017-11-27:
@smlng I do not see the behaviour from 4 when adding the entry as soon as the RIOT instance is started (so omitting step 1 and 2). I have not yet tested the full steps of your comment yet as #8144 is not yet merged.

smlng, 2017-11-27:
@bergzand you're right, works - I made a mistake: I added the entry manually **but** for the wrong interface, i.e. 7 instead of 6 -> with 6lo the interface pid is 7, but w/o 6lo (as for ethernet/tap on native) the pid is 6 😦.

---

Interesting side effect: after manually adding a faulty entry and ping6 on the host lladdr there are 2 entries in the nib for the same address and lladdr, but on different interfaces. While the `nib neigh add` commands _knows_ interfaces `del` does not, and you can only delete the first matching entry in the NIB for the requested address. Which raises 2 questions: 

- should we extend `nib neigh del` to _know_ interfaces, too?
- is it correct (or should it be possible) two have 2 entries for the same IP and LLaddr but on different interfaces - IMHO it doesn't make sense, or does it?

smlng, 2017-11-27:
>  you can only delete the first matching entry in the NIB 

or more correct: you have to delete in a loop to remove all entries, but you cannot select (or know) which will be deleted first - at least if you do not know in which order they were added 😉

smlng, 2017-11-27:
I started to test on boards, first issue/question: with 6lo active RFC6775 applies, which says ND is host-initiated and does not use multicast. So question for latter: why does RIOT still join its sol. mcast address? Here `ifconfig` for PhyNode (pba-d-01-kw2x):

```
> ifconfig
2017-11-27 12:14:57,961 - INFO #  ifconfig
2017-11-27 12:14:57,966 - INFO # Iface  7  HWaddr: 13:36  Channel: 26  NID: 0x23
2017-11-27 12:14:57,970 - INFO #           Long HWaddr: d3:c1:6d:73:ab:43:13:36 
2017-11-27 12:14:57,973 - INFO #            TX-Power: 0dBm  State: IDLE 
2017-11-27 12:14:57,976 - INFO #           ACK_REQ  AUTOCCA
2017-11-27 12:14:57,979 - INFO #           MTU:1280  HL:hu  RTR  IPHC  
2017-11-27 12:14:57,982 - INFO #           Source address length: 8
2017-11-27 12:14:57,985 - INFO #           Link type: wireless
2017-11-27 12:14:58,000 - INFO #           inet6 addr: fe80::d1c1:6d73:ab43:1336  scope: local  VAL
2017-11-27 12:14:58,001 - INFO #           inet6 group: ff02::2
2017-11-27 12:14:58,001 - INFO #           inet6 group: ff02::1
2017-11-27 12:14:58,002 - INFO #           inet6 group: ff02::1:ff43:1336
2017-11-27 12:14:58,003 - INFO #           inet6 group: ff02::1a
2017-11-27 12:14:58,003 - INFO #           
2017-11-27 12:14:58,006 - INFO #           Statistics for Layer 2
2017-11-27 12:14:58,009 - INFO #             RX packets 0  bytes 0
2017-11-27 12:14:58,014 - INFO #             TX packets 3 (Multicast: 3)  bytes 86
2017-11-27 12:14:58,017 - INFO #             TX succeeded 2 errors 0
2017-11-27 12:14:58,019 - INFO #           Statistics for IPv6
2017-11-27 12:14:58,022 - INFO #             RX packets 0  bytes 0
2017-11-27 12:14:58,027 - INFO #             TX packets 3 (Multicast: 3)  bytes 178
2017-11-27 12:14:58,030 - INFO #             TX succeeded 3 errors 0
2017-11-27 12:14:58,030 - INFO #
```

miri64, 2017-11-27:
`gnrc_networking` is in 6LR configuration, RFC 6775 to optionally do [classic NDP between routers](https://tools.ietf.org/html/rfc6775#section-6.5.5) and so we do. If you want the node to be in host configuration (6LN) either remove RPL and the `_router` in `gnrc_ipv6_router_default` or use the `gcoap` example.

Can you please summarize your findings again? I'm a bit confused what the issue is still.

miri64, 2017-11-27:
> - should we extend `nib neigh del` to _know_ interfaces, too?
> - is it correct (or should it be possible) ~~two~~ [to] have 2 entries for the same IP and LLaddr but on different interfaces - IMHO it doesn't make sense, or does it?

Is this the only remaining issue in this issue (wording chosen intentionally ;-P)?

Yes it does make sense to have 2 entries with the same *IP* address at least (lladdr might also work if they are in distinct networks). An example: A device has two radios and is in direct neighborhood for the border routers of each radio network. The administrators for both border routers are a bit lazy, so for both the link-local address is `fe80::1`. Since they are link-local addresses the interface needs to be known anyways so this is possible (and since the distinction in the NIB is done as an (addr, iface)-tuple it is there as well).

Regarding the `nib neigh del`: Yes it does make sense when you put it this way. Will fix.

cgundogan, 2017-11-27:
Not a blocking issue, but important for the long run:
My setup is:
```
        6lbr (RasPi with Radvd)
6lr (gnrc_networking)     6lh (gcoap)
```
All nodes are in range.

The 6lh sets a default route to the router from the first received RA.
This would lead to a ping (6lh -> 6lbr) scenario, where 6lh sends a ping request to 6lbr via 6lr and the ping reply is sent back directly from 6lbr to 6lh. I would expect that the ping request is sent directly from 6lh to 6lbr.

After some research, we came across [RFC4191](https://tools.ietf.org/html/rfc4191). RAs can basically contain a preference number and according to [RFC6775](https://tools.ietf.org/html/rfc6775#section-6), the preference MUST be set to high, if this preference option is used. IMO it makes sense to choose the 6lbr directly in a network where severel 6lrs are present in the vicinity.

I was able to set the preference of the 6lbr (via radvd) to high, but RIOT is not parsing that at the moment.

miri64, 2017-11-27:
@cgundogan can you open a separate issue for that (though I added the functionality in RFC4191 to #3052). However, I don't think RFC4191 alone will fix this issue, but rather that is also an interplay with the issue @smlng pointed out during today's call, that (current) Linux does not behave according to RFC6775 and expects the neighbor to be in the solicited-nodes address of its address. Maybe @alexaring can give a hint on the status of 6Lo-ND in Linux, if he is still up-to-date on that.

miri64, 2017-12-11:
@cgundogan did you open the issue I asked for? If not, please do so. Since #7925 is merged now, I'm closing this issue.

closed 2017-12-11 by miri64

alexaring, 2017-12-11:
@miri64 there is not support right now, except I did some user space implementation for sending 6CO and parsing them (then setup contexts afterwards), just early state and RIOT had already support for this RFC6775 "feature".

I am aware of some contiki fork for RFC6775 (which they didn't had upstream yet, don't know the current state) - don't know the name/repository anymore. Sorry.

Linux support needs some time do find some "acceptable" architecture of "what doing in kernel and what in user space", which also fits to the current IPv6 architecture.

Maybe @Stefan-Schmidt knows something about that... People are more interessted in Thread, but they don't use RFC6775 at all (so far I saw).